ylwhxht/L4DR #2: fog simulation

fightingand, 2024-12-18:
Hello,I need  the fog simulation on the VoD lidar point cloud, can  you send it to my email wuyl@mails.qust.edu.cn ?Thanks!

ylwhxht, 2024-12-18:
> email

Okay, but it may require some time, I need to upload it because it's too large to send by email. I expect to complete it within two days.

ylwhxht, 2024-12-18:
> Hello,I need the fog simulation on the VoD lidar point cloud, can you send it to my email [wuyl@mails.qust.edu.cn](mailto:wuyl@mails.qust.edu.cn) ?Thanks!

Okay, but it may require some time, I need to upload it because it's too large to send by email. I expect to complete it within two days.

fightingand, 2024-12-18:
Thanks very much. I also need the pretrained models.

ylwhxht, 2024-12-18:
> Thanks very much. I also need the pretrained models.

This is the link to the pre-trained model (valid for 7 days). 
https://pan.baidu.com/s/1j_JW8zNRvuGc0UNi1lnHmw?pwd=s0tr (code = s0tr)

_Please note: There may be some differences in the AP performance between this .pth and paper, as I accidentally couldn't find the checkpoint that was reported in the paper. But the checkpoint accuracy of this version is also similar. However, due to too many evaluation settings in our experiment (different classes, different fog levels, and different difficulties), this may lead to differences in some specific results (good or bad), but the total performance is similar. You can first refer to the accuracy of this model or try retraining it yourself._

The dataset is still being uploaded and is expected to be completed tomorrow.

fightingand, 2024-12-19:
我按照您的项目首先复现了一下InterFusion，在单个A100GPU上训练了80轮，batch_size=8，得出的结果是{'entire_area': {'Car_3d_all': '50.9', 'Pedestrian_3d_all': '46.62', 'Cyclist_3d_all': '72.94', '3d_all': '56.82'}, 'roi': {'Car_3d_all': '79.49', 'Pedestrian_3d_all': '55.48', 'Cyclist_3d_all': '88.66', '3d_all': '74.55'}}，其中Car和Pedestrian类别与论文出入比较大，请问大概是什么原因呢

ylwhxht, 2024-12-19:
> 我按照您的项目首先复现了一下InterFusion，在单个A100GPU上训练了80轮，batch_size=8，得出的结果是{'entire_area': {'Car_3d_all': '50.9', 'Pedestrian_3d_all': '46.62', 'Cyclist_3d_all': '72.94', '3d_all': '56.82'}, 'roi': [...]

我可能也没办法直接定位问题，我提供一下InterFusion的模型与训练日志供你排查，建议可以尝试运行InterFusion模型是否能得到对应的结果来检查能否正常测试。
通过百度网盘分享的文件：InterFusion
链接：https://pan.baidu.com/s/1TBfMmd4aYc8ShGsOQ5tS_g?pwd=nx1u 
提取码：nx1u

ylwhxht, 2024-12-19:
> 我按照您的项目首先复现了一下InterFusion，在单个A100GPU上训练了80轮，batch_size=8，得出的结果是{'entire_area': {'Car_3d_all': '50.9', 'Pedestrian_3d_all': '46.62', 'Cyclist_3d_all': '72.94', '3d_all': '56.82'}, 'roi': [...]

另外，InterFusion本质就是PointPillar加上一点模块。
建议从最简单的LiDAR-based PointPillar复现

fightingand, 2024-12-19:
您的：
2024-03-19 00:46:21,325   INFO  Database filter by min points Car: 15608 => 8604
2024-03-19 00:46:21,329   INFO  Database filter by min points Pedestrian: 16143 => 8557
2024-03-19 00:46:21,330   INFO  Database filter by min points Cyclist: 6685 => 4964
我的：
2024-12-18 21:40:30,982   INFO  Database filter by min points Car: 15608 => 7360
2024-12-18 21:40:30,984   INFO  Database filter by min points Pedestrian: 16143 => 908
2024-12-18 21:40:30,985   INFO  Database filter by min points Cyclist: 6685 => 221
应该是这里出现了问题，但是我的数据集的处理是完全按照您的配置文件进行的
          PREPARE: {
             filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5'],
             filter_by_difficulty: [-1],
          }

ylwhxht, 2024-12-19:
> 您的： 2024-03-19 00:46:21,325 INFO Database filter by min points Car: 15608 => 8604 2024-03-19 00:46:21,329 INFO Database filter by min points Pedestrian: 16143 => 8557 2024-03-19 00:46:21,330 INFO [...]

很细致的检查，可能是数据处理的代码出现了一点问题，我check一下

ylwhxht, 2024-12-19:
> 您的： 2024-03-19 00:46:21,325 INFO Database filter by min points Car: 15608 => 8604 2024-03-19 00:46:21,329 INFO Database filter by min points Pedestrian: 16143 => 8557 2024-03-19 00:46:21,330 INFO [...]

我又尝试使用clone的代码重新生成了一次，还是得到了跟之前一致的结果。

这个问题主要是由于生成gt_base的时候lidar_num_points_in_gt和radar_num_points_in_gt（[dataset](https://github.com/ylwhxht/L4DR/blob/main/pcdet/datasets/vod/vod_dataset.py) 中的create_groundtruth_database方法）不对，导致大量gt点数过少被过滤掉影响到了数据增强。

因为我重新生成了一遍，基本上确保了代码没问题，大概率数据组织可能存在一定问题，建议重新检查一下组织rlfusion_5f文件夹是否存在错误。

你也可以在执行`python -m pcdet.datasets.vod.vod_dataset create_vod_infos tools/cfgs/dataset_configs/Vod_fusion.yaml`的过程输出一下`print(db_info['lidar_num_points_in_gt'],db_info['radar_num_points_in_gt'])`，查看他们的点数是否出现问题，进而往前定位到问题出在哪。

fightingand, 2024-12-19:
谢谢，我更新一下数据集试试

ylwhxht, 2024-12-21:
> 谢谢，我更新一下数据集试试

> Hello,I need the fog simulation on the VoD lidar point cloud, can you send it to my email [wuyl@mails.qust.edu.cn](mailto:wuyl@mails.qust.edu.cn) ?Thanks!

这是[雾模拟的VoD lidar点云](https://huggingface.co/datasets/hx24/VoD_fog_simulation/tree/main)，权限因素仅开放7天尽快下载

closed 2024-12-21 by ylwhxht

Mangonn, 2025-03-20:
@ylwhxht May I request the pretrained model? Thanks for your time！

ylwhxht, 2025-03-20:
> [@ylwhxht](https://github.com/ylwhxht) 您好，我也想要一下pre-trained model可以吗，谢谢！
VoD数据集的还是Kradar数据集的

Mangonn, 2025-03-20:
> > [@ylwhxht](https://github.com/ylwhxht) 您好，我也想要一下pre-trained model可以吗，谢谢！
> 
> 你好，是要VoD数据集的还是Kradar数据集的？

想要VoD的，Kradar在别的回复里找到啦

Mangonn, 2025-03-20:
> > [@ylwhxht](https://github.com/ylwhxht) 您好，我也想要一下pre-trained model可以吗，谢谢！
> 
> 你好，是要VoD数据集的还是Kradar数据集的？

想要VoD的，谢谢！！！！！

Mangonn, 2025-03-20:
@ylwhxht  想要VoD数据集的，谢谢～

ylwhxht, 2025-03-20:
> [@ylwhxht](https://github.com/ylwhxht) 想要VoD数据集的，谢谢～

链接: https://pan.baidu.com/s/1RRhByabe-D2gLp2cYEDj_Q?pwd=61vm 提取码: 61vm 😊