ylwhxht/L4DR #4: training on fog simulation datasets

17BuGs, 2025-02-10:
Dear author,

我尝试使用您提供的fog simulation数据集来训练L4DR，但是在验证集上的效果与论文中给出的指标不能很好地对应。我在`vod_dataset.py`中修改了`self.use_fog=1`，`self.fog_I=4`，并将`self.sim_info_path`替换为`_CVL_beta_0.200`。这是我在`fog level = 4`条件下训练出的模型指标：

> Car AP@0.70, 0.70, 0.70:
> bbox AP:36.2393, 37.5138, 33.1076
> bev  AP:41.6963, 38.3321, 34.4460
> 3d   AP:23.3377, 23.6655, 20.6604
> aos  AP:33.75, 35.98, 31.64
> Car AP_R40@0.70, 0.70, 0.70:
> bbox AP:33.0580, 35.1609, 30.0094
> bev  AP:39.5321, 36.4095, 31.3007
> 3d   AP:18.7568, 18.9421, 16.2214
> aos  AP:30.33, 33.44, 28.50
> Car AP@0.70, 0.50, 0.50:
> bbox AP:36.2393, 37.5138, 33.1076
> bev  AP:56.5059, 48.1610, 43.0019
> 3d   AP:43.0323, 42.7390, 35.2223
> aos  AP:33.75, 35.98, 31.64
> Car AP_R40@0.70, 0.50, 0.50:
> bbox AP:33.0580, 35.1609, 30.0094
> bev  AP:56.2651, 46.5806, 39.4289
> 3d   AP:40.9279, 39.1024, 33.8743
> aos  AP:30.33, 33.44, 28.50
> Pedestrian AP@0.50, 0.50, 0.50:
> bbox AP:44.9698, 44.3642, 39.7489
> bev  AP:37.1579, 36.6389, 33.5584
> 3d   AP:28.7844, 28.5246, 25.5302
> aos  AP:43.15, 41.50, 37.18
> Pedestrian AP_R40@0.50, 0.50, 0.50:
> bbox AP:42.8861, 42.6435, 38.2201
> bev  AP:34.2510, 33.5144, 30.0521
> 3d   AP:25.3636, 24.4102, 21.7367
> aos  AP:40.97, 39.41, 35.21
> Pedestrian AP@0.50, 0.25, 0.25:
> bbox AP:44.9698, 44.3642, 39.7489
> bev  AP:57.8001, 56.0685, 49.8582
> 3d   AP:54.4795, 53.0960, 47.2407
> aos  AP:43.15, 41.50, 37.18
> Pedestrian AP_R40@0.50, 0.25, 0.25:
> bbox AP:42.8861, 42.6435, 38.2201
> bev  AP:58.0170, 54.9036, 49.7708
> 3d   AP:53.4446, 51.1579, 45.2932
> aos  AP:40.97, 39.41, 35.21
> Cyclist AP@0.50, 0.50, 0.50:
> bbox AP:62.6840, 60.4740, 54.3611
> bev  AP:57.0309, 52.4074, 46.2434
> 3d   AP:38.9169, 35.5485, 33.8006
> aos  AP:62.51, 60.23, 54.15
> Cyclist AP_R40@0.50, 0.50, 0.50:
> bbox AP:64.0400, 58.9724, 53.9712
> bev  AP:55.2764, 50.7206, 45.9953
> 3d   AP:36.4385, 34.1779, 30.4764
> aos  AP:63.83, 58.72, 53.73
> Cyclist AP@0.50, 0.25, 0.25:
> bbox AP:62.6840, 60.4740, 54.3611
> bev  AP:80.3171, 74.2547, 66.6210
> 3d   AP:79.6067, 73.3283, 65.9682
> aos  AP:62.51, 60.23, 54.15
> Cyclist AP_R40@0.50, 0.25, 0.25:
> bbox AP:64.0400, 58.9724, 53.9712
> bev  AP:80.5271, 75.8326, 68.8771
> 3d   AP:79.5810, 73.6063, 66.7312
> aos  AP:63.83, 58.72, 53.73

请问您是否可以指导一下问题出在哪里？

感谢！

ylwhxht, 2025-02-10:
你好，
感谢你对我们工作的关注。
这个精度确实不太合理，我推荐从以下方面去排查：

1. 可视化一下CVL_200的点云，检查是否是对应的雾强度（如果是直接下载的则跳过）。
2. 在vod_dataset.py文件下的get_pc函数对应位置输出读取的lidar_file路径，是否真的使用了CVL_200的点云。
3. 训练的时候不要使用use_fog=1(fog_I没什么用不用管)，因为对特定强度的雾专门再训练本质上是一种数据泄露，不能说为了测试fog_level=4的性能专门只在fog_level=4的数据上训练。不管在什么fog_level上测试，训练始终都是use_fog=2采用随机采样雾模拟的方式进行训练。仅在测试的时候才调整为对应的fog_level。

17BuGs, 2025-02-11:
感谢您提供的建议！

我在训练的时候确实错把use_fog设置成1。不过经过修改之后，训练得到的指标仍然不合理。我不确定在数据增强的配置中是否应该使用gt_sampling，但是经过测试，问题可能主要不是在这里。下面是我使用的配置文件，能否麻烦您帮忙check一下呢？

```yaml
CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/Vod_fusion.yaml
    POINT_CLOUD_RANGE: [0, -25.6, -3, 51.2, 25.6, 2]
    VOD_EVA : False
    DATA_PROCESSOR:
        - NAME: mask_points_and_boxes_outside_range
          REMOVE_OUTSIDE_BOXES: True
        - NAME: sample_points
          NUM_POINTS: {
                'train': 2048,
                'test': 2048
            }
        - NAME: shuffle_points
          SHUFFLE_ENABLED: {
            'train': True,
            'test': False
          }
        
        - NAME: transform_points_to_voxels
          VOXEL_SIZE: [0.16, 0.16, 4]
          MAX_POINTS_PER_VOXEL: 32
          MAX_NUMBER_OF_VOXELS: {
            'train': 16000,
            'test': 40000
          }
          
    DATA_AUGMENTOR:
        DISABLE_AUG_LIST: ['placeholder', 'gt_sampling']
        AUG_CONFIG_LIST:
            # - NAME: gt_sampling
            #   USE_ROAD_PLANE: False
            #   DB_INFO_PATH:
            #       - vod_dbinfos_train.pkl
            #   PREPARE: {
            #      filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5'],
            #      #filter_by_difficulty: [-1],
            #   }

            #   SAMPLE_GROUPS: ['Car:15','Pedestrian:15', 'Cyclist:15']
            #   NUM_POINT_FEATURES_L: 4
            #   NUM_POINT_FEATURES_R: 7
            #   DATABASE_WITH_FAKELIDAR: False
            #   REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
            #   LIMIT_WHOLE_SCENE: False

            - NAME: random_world_flip
              ALONG_AXIS_LIST: ['x']

            - NAME: random_world_rotation
              WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

            - NAME: random_world_scaling
              WORLD_SCALE_RANGE: [0.95, 1.05]

MODEL:
    NAME: PointPillarMask

    VFE:
        NAME: MME_PillarVFE
        WITH_DISTANCE: False
        USE_ABSLOTE_XYZ: True
        USE_XYZ: True
        USE_RCS: True
[... 127 lines cut ...]
            NMS_THRESH: 0.01
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 8
    NUM_EPOCHS: 30

    OPTIMIZER: adam_onecycle
    LR: 0.001
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10
```

ylwhxht, 2025-02-11:
> 训练得到的指标仍然不合理

“训练得到的指标仍然不合理”具体是什么情况呢

ylwhxht, 2025-02-11:
> > > 训练得到的指标仍然不合理
> > 
> > 
> > “训练得到的指标仍然不合理”具体是什么情况呢
> 
> 得到的精度还是比较高，在fog_level=4的指标如下：
> 
> > Car AP@0.70, 0.70, 0.70:
> > bbox AP:33.6159, 33.7003, 30.2244
> > bev  AP:36.5682, 33.8911, 29.6338
> > 3d   AP:22.1185, 23.5180, 19.3173
> > aos  AP:30.90, 31.59, 28.27
> > Car AP_R40@0.70, 0.70, 0.70:
> > bbox AP:30.9456, 31.7752, 26.8488
> > bev  AP:33.6346, 32.1916, 26.2202
> > 3d   AP:18.8591, 19.1371, 15.3389
> > aos  AP:28.14, 29.38, 24.75
> > Car AP@0.70, 0.50, 0.50:
> > bbox AP:33.6159, 33.7003, 30.2244
> > bev  AP:47.9200, 43.3712, 39.3651
> > 3d   AP:39.3789, 38.8155, 33.9216
> > aos  AP:30.90, 31.59, 28.27
> > Car AP_R40@0.70, 0.50, 0.50:
> > bbox AP:30.9456, 31.7752, 26.8488
> > bev  AP:47.4496, 43.0906, 36.2570
> > 3d   AP:37.6611, 36.1561, 31.1104
> > aos  AP:28.14, 29.38, 24.75
> > Pedestrian AP@0.50, 0.50, 0.50:
> > bbox AP:39.0465, 37.0344, 32.0456
> > bev  AP:29.7975, 29.0255, 25.8505
> > 3d   AP:23.5751, 21.8424, 20.0924
> > aos  AP:37.42, 34.34, 29.79
> > Pedestrian AP_R40@0.50, 0.50, 0.50:
> > bbox AP:37.3627, 34.1645, 30.1424
> > bev  AP:27.0301, 25.3476, 21.9890
> > 3d   AP:19.5265, 17.9358, 15.1412
> > aos  AP:35.40, 31.15, 27.42
> > Pedestrian AP@0.50, 0.25, 0.25:
> > bbox AP:39.0465, 37.0344, 32.0456
> > bev  AP:53.5940, 47.7856, 41.9791
> > 3d   AP:48.4577, 45.0803, 39.5140
> > aos  AP:37.42, 34.34, 29.79
> > Pedestrian AP_R40@0.50, 0.25, 0.25:
> > bbox AP:37.3627, 34.1645, 30.1424
> > bev  AP:51.6437, 45.8675, 41.0616
> > 3d   AP:46.6607, 42.0832, 37.4864
> > aos  AP:35.40, 31.15, 27.42
> > Cyclist AP@0.50, 0.50, 0.50:
> > bbox AP:58.3277, 54.0592, 51.1167
> > bev  AP:40.6639, 39.8894, 35.0662
> > 3d   AP:30.3685, 29.5596, 26.3338
> > aos  AP:58.16, 53.82, 50.90
> > Cyclist AP_R40@0.50, 0.50, 0.50:
> > bbox AP:56.7674, 54.3014, 48.5384
> > bev  AP:38.4507, 37.1106, 33.2699
> > 3d   AP:26.4225, 25.2034, 22.3228
> > aos  AP:56.59, 54.03, 48.29
> > Cyclist AP@0.50, 0.25, 0.25:
> > bbox AP:58.3277, 54.0592, 51.1167
> > bev  AP:74.4035, 72.0808, 65.2240
> > 3d   AP:71.8576, 65.5416, 62.7069
> > aos  AP:58.16, 53.82, 50.90
> > Cyclist AP_R40@0.50, 0.25, 0.25:
> > bbox AP:56.7674, 54.3014, 48.5384
> > bev  AP:77.1525, 71.3517, 64.4253
> > 3d   AP:71.8723, 67.6285, 60.8577
> > aos  AP:56.59, 54.03, 48.29

你是否修改了用于训练随机采样的强度self.sim_info_path_list = [
            Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.005/'),
            Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.010/'),
            Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.020/'),
            Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.030/'),
        ]

17BuGs, 2025-02-12:
> 你是否修改了用于训练随机采样的强度self.sim_info_path_list = [
> Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.005/'),
> Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.010/'), [...]

是的，现在修改之后的内容是这样：

```python
        self.use_fog = 2    # 0
        # train = 2
        self.fog_I = 4  # 0

        self.train = (self.split != 'val')
        # 0 no fog 
        # 1 specific fog 
        # 2 random fog
        
        self.sim_info_path_list = [
            Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.030/'),
            Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.060/'),
            Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.100/'),
            Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.200/'),
        ]
        self.sim_info_path = Path('/home/icdm/wf/L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.200/') 
```

如果方便的话，您可以提供一份训练日志以便我定位一下问题吗？

十分感谢！

ylwhxht, 2025-02-12:
> > 你是否修改了用于训练随机采样的强度self.sim_info_path_list = [
> > Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.005/'),
> > Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.010/'),
> > Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.020/'),
> > Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.030/'),
> > ]
> 
> 是的，现在修改之后的内容是这样：
> 
>         self.use_fog = 2    # 0
>         # train = 2
>         self.fog_I = 4  # 0
> 
>         self.train = (self.split != 'val')
>         # 0 no fog 
>         # 1 specific fog 
>         # 2 random fog
>         
>         self.sim_info_path_list = [
>             Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.030/'),
>             Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.060/'),
>             Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.100/'),
>             Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.200/'),
>         ]
>         self.sim_info_path = Path('/home/icdm/wf/L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.200/') 
> 如果方便的话，您可以提供一份训练日志以便我定位一下问题吗？
> 
> 十分感谢！

> > 你是否修改了用于训练随机采样的强度self.sim_info_path_list = [
> > Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.005/'),
> > Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.010/'),
> > Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.020/'),
> > Path('/mnt/32THHD/view_of_delft_PUBLIC/fog_sim_lidar/_CVL_beta_0.030/'),
> > ]
> 
> 是的，现在修改之后的内容是这样：
> 
>         self.use_fog = 2    # 0
>         # train = 2
>         self.fog_I = 4  # 0
> 
>         self.train = (self.split != 'val')
>         # 0 no fog 
>         # 1 specific fog 
>         # 2 random fog
>         
>         self.sim_info_path_list = [
>             Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.030/'),
>             Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.060/'),
>             Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.100/'),
>             Path('path to L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.200/'),
>         ]
>         self.sim_info_path = Path('/home/icdm/wf/L4DR/data/vod/fog_sim_lidar/_CVL_beta_0.200/') 
> 如果方便的话，您可以提供一份训练日志以便我定位一下问题吗？
> 
> 十分感谢！

这个就是问题了啊，用于训练的数据不一样结果差自然比较大。我们用您修改前的那几个强度进行训练，主要测试大量未知域上的结果。这是我们的设置，如果你需要跟我们汇报的精度相对一致就需要保证用于训练的数据范围是一致的（也就是未修改的版本）。
当然您也可以就按自己这个设置来，使用复现的精度进行对比，只要说明即可。

17BuGs, 2025-02-12:
> 这个就是问题了啊，用于训练的数据不一样结果差自然比较大。我们用您修改前的那几个强度进行训练，主要测试大量未知域上的结果。这是我们的设置，如果你需要跟我们汇报的精度相对一致就需要保证用于训练的数据范围是一致的（也就是未修改的版本）。
> 当然您也可以就按自己这个设置来，使用复现的精度进行对比，只要说明即可。

好的好的，我理解了。非常感谢您的回复！

closed 2025-02-18 by ylwhxht