ylwhxht/L4DR #5: kitti评估指标

fightingand, 2025-02-18:
论文中的结果表明IOU为0.5，0.25，0.25
但是eval代码则为：
    overlap_0_7 = np.array([[0.7, 0.5, 0.5, 0.7,
                             0.5, 0.7], [0.7, 0.5, 0.5, 0.7, 0.5, 0.7],
                            [0.7, 0.5, 0.5, 0.7, 0.5, 0.7]])
    overlap_0_5 = np.array([[0.7, 0.5, 0.5, 0.7,
                             0.5, 0.5], [0.5, 0.25, 0.25, 0.5, 0.25, 0.5],
                            [0.5, 0.25, 0.25, 0.5, 0.25, 0.5]])
是否有问题？

ylwhxht, 2025-02-18:
> overlap

没有问题，这里使用的就是kitti的官方配置。
实际上这里的3x6行代表的是3种评估类型(bbox,bev,3d)、6个类别的阈值
我们汇报的是3D的精度，实际上就是最后一行，把overlap_0_5 矩阵前三列（0: 'Car',1: 'Pedestrian',2: 'Cyclist'）阈值拿出来就是[0.5,0.25,0.25]。

实际上，最终评估的输出样例为，我们选取的就是[0.5,0.25,0.25]阈值下、AP@40、3d类型的精度（以下加粗部分）：
Car AP@0.70, 0.70, 0.70:
bbox AP:73.7505, 69.9075, 65.4887
bev  AP:72.3862, 67.9770, 60.8954
3d   AP:54.7070, 50.9549, 45.9244
aos  AP:66.66, 66.91, 62.56
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:74.3846, 70.4607, 64.6048
bev  AP:72.4947, 67.7413, 61.8725
3d   AP:53.2550, 50.9648, 44.3382
aos  AP:66.50, 67.15, 61.45
Car AP@0.70, 0.50, 0.50:
bbox AP:73.7505, 69.9075, 65.4887
bev  AP:86.4600, 79.4332, 75.7536
3d   AP:82.7627, 75.9137, 69.2504
aos  AP:66.66, 66.91, 62.56
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:74.3846, 70.4607, 64.6048
bev  AP:88.4988, 81.1465, 75.4193
**3d   AP:84.5420, 76.2586, 68.9325**
aos  AP:66.50, 67.15, 61.45
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:70.7758, 68.7821, 62.8140
bev  AP:63.7429, 61.1565, 55.0022
3d   AP:58.1607, 54.0307, 47.3740
aos  AP:67.60, 63.81, 58.06
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:71.4784, 68.7035, 62.3435
bev  AP:64.6286, 60.6164, 54.1497
3d   AP:57.8897, 53.0833, 46.8388
aos  AP:67.98, 63.27, 57.12
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:70.7758, 68.7821, 62.8140
bev  AP:71.9941, 69.9392, 63.8805
3d   AP:71.9059, 69.6978, 63.7208
aos  AP:67.60, 63.81, 58.06
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:71.4784, 68.7035, 62.3435
bev  AP:72.7852, 70.5262, 64.2541
**3d   AP:72.7147, 70.3924, 64.1054**
aos  AP:67.98, 63.27, 57.12
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:87.8901, 87.1684, 78.7944
bev  AP:87.7284, 86.8523, 78.4172
3d   AP:86.3945, 85.0170, 77.1408
aos  AP:87.68, 86.94, 78.58
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:93.4795, 89.7642, 82.3998
bev  AP:93.2498, 89.3240, 81.4812
3d   AP:91.6257, 86.3702, 78.4017
aos  AP:93.25, 89.51, 82.14
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:87.8901, 87.1684, 78.7944
bev  AP:92.7042, 87.5044, 83.1941
3d   AP:92.6869, 87.5044, 83.1671
aos  AP:87.68, 86.94, 78.58
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:93.4795, 89.7642, 82.3998
bev  AP:94.9063, 90.1241, 84.2299
**3d   AP:94.8960, 90.1172, 84.0917**
aos  AP:93.25, 89.51, 82.14

ylwhxht, 2025-02-18:
> 论文中的结果表明IOU为0.5，0.25，0.25 但是eval代码则为： overlap_0_7 = np.array([[0.7, 0.5, 0.5, 0.7, 0.5, 0.7], [0.7, 0.5, 0.5, 0.7, 0.5, 0.7], [0.7, 0.5, 0.5, 0.7, 0.5, 0.7]]) overlap_0_5 = np.array([[0.7, 0.5, 0.5, 0.7, 0.5, 0.5], [0.5, 0.25, 0.25, 0.5, 0.25, 0.5], [0.5, 0.25, 0.25, 0.5, 0.25, 0.5]]) 是否有问题？

没有问题，这里使用的就是kitti的官方配置。
实际上这里的3x6行代表的是3种评估类型(bbox,bev,3d)、6个类别的阈值
我们汇报的是3D的精度，实际上就是最后一行，把overlap_0_5 矩阵前三列（0: 'Car',1: 'Pedestrian',2: 'Cyclist'）阈值拿出来就是[0.5,0.25,0.25]。

实际上，最终评估的输出样例为，我们选取的就是[0.5,0.25,0.25]阈值下、AP@40、3d类型的精度（以下加粗部分）：
Car AP@0.70, 0.70, 0.70:
bbox AP:73.7505, 69.9075, 65.4887
bev  AP:72.3862, 67.9770, 60.8954
3d   AP:54.7070, 50.9549, 45.9244
aos  AP:66.66, 66.91, 62.56
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:74.3846, 70.4607, 64.6048
bev  AP:72.4947, 67.7413, 61.8725
3d   AP:53.2550, 50.9648, 44.3382
aos  AP:66.50, 67.15, 61.45
Car AP@0.70, 0.50, 0.50:
bbox AP:73.7505, 69.9075, 65.4887
bev  AP:86.4600, 79.4332, 75.7536
3d   AP:82.7627, 75.9137, 69.2504
aos  AP:66.66, 66.91, 62.56
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:74.3846, 70.4607, 64.6048
bev  AP:88.4988, 81.1465, 75.4193
**3d   AP:84.5420, 76.2586, 68.9325**
aos  AP:66.50, 67.15, 61.45
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:70.7758, 68.7821, 62.8140
bev  AP:63.7429, 61.1565, 55.0022
3d   AP:58.1607, 54.0307, 47.3740
aos  AP:67.60, 63.81, 58.06
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:71.4784, 68.7035, 62.3435
bev  AP:64.6286, 60.6164, 54.1497
3d   AP:57.8897, 53.0833, 46.8388
aos  AP:67.98, 63.27, 57.12
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:70.7758, 68.7821, 62.8140
bev  AP:71.9941, 69.9392, 63.8805
3d   AP:71.9059, 69.6978, 63.7208
aos  AP:67.60, 63.81, 58.06
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:71.4784, 68.7035, 62.3435
bev  AP:72.7852, 70.5262, 64.2541
**3d   AP:72.7147, 70.3924, 64.1054**
aos  AP:67.98, 63.27, 57.12
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:87.8901, 87.1684, 78.7944
bev  AP:87.7284, 86.8523, 78.4172
3d   AP:86.3945, 85.0170, 77.1408
aos  AP:87.68, 86.94, 78.58
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:93.4795, 89.7642, 82.3998
bev  AP:93.2498, 89.3240, 81.4812
3d   AP:91.6257, 86.3702, 78.4017
aos  AP:93.25, 89.51, 82.14
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:87.8901, 87.1684, 78.7944
bev  AP:92.7042, 87.5044, 83.1941
3d   AP:92.6869, 87.5044, 83.1671
aos  AP:87.68, 86.94, 78.58
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:93.4795, 89.7642, 82.3998
bev  AP:94.9063, 90.1241, 84.2299
**3d   AP:94.8960, 90.1172, 84.0917**
aos  AP:93.25, 89.51, 82.14

closed 2025-02-18 by ylwhxht

fightingand, 2025-02-18:
好的，非常感谢