ylwhxht/L4DR #6: 可视化

15563264286, 2025-02-19:
可以详细说说可视化吗，我不太明白如何同时将点云和雷达点可视化到同一张图片上

15563264286, 2025-02-24:
我按照您的步骤尝试了一下，但一直不成功，能否提供一下您的可视化工具代码供我排查一下错误呢？

ylwhxht, 2025-02-24:
> 我按照您的步骤尝试了一下，但一直不成功，能否提供一下您的可视化工具代码供我排查一下错误呢？

我的可视化工具就是上面回答的那些了，推理的时候调用save_frame_vis保存vis为pickle文件，在提供可视化工具代码库上load就是了。

你不成功的原因/具体体现是什么呢？

15563264286, 2025-02-24:
可视化工具代码库上似乎并未提供kitti数据集的pkl文件加载，只有txt加载

ylwhxht, 2025-02-24:
> 可视化工具代码库上似乎并未提供kitti数据集的pkl文件加载，只有txt加载

自己实现就可以了呀，load pkl文件很简单的，核心是得到数据之后使用他的可视化。
以下代码供参考，注意实际使用的需要对齐pickle文件的key，只要根据自己保存时候的字典进行调整即可。代码很粗糙，建议重点看代码的调用逻辑然后自己实现:
```
from sys import implementation
from viewer.viewer import Viewer
import numpy as np
from dataset.waymo_base import WaymoDataset
import os
from tqdm import tqdm
import random
from load_pcd import load_pcd_data
import torch
import pickle
len1 = 0
len2 = 0
from viewer.box_op import get_points_in_boxes3d
class_names = ['Vehicle', 'Pedestrian', 'Cyclist']
l_path = r'D:\Study\papers\AAAI2025\vis\l_vis_all100.pkl'
l_vis_list = r_vis_list = lr_vis_list =  None
if os.path.exists(l_path):
    with open(l_path, 'rb') as f:
        l_vis_list = pickle.load(f)
    len1 = len(l_vis_list)

r_path = r'C:\Users\ylwhxht\Desktop\r_vis.pkl'
if os.path.exists(r_path):
    with open(r_path, 'rb') as f:
        r_vis_list = pickle.load(f)
    len1 = len(r_vis_list)
vi = Viewer(bg=(245,245,245))

lr_path = r"D:\Study\papers\AAAI2025\vis\Inter_vis_all100.pkl"
if os.path.exists(lr_path):
    with open(lr_path, 'rb') as f:
        lr_vis_list = pickle.load(f)
    len1 = len(lr_vis_list)
vi = Viewer(bg=(245,245,245))

def gt_mask(gt, pc):
    t = get_points_in_boxes3d(pc, gt)
    return gt[t>4]
    
for i in range(312,len1):
    l_vis = r_vis = None
    if l_vis_list is not None:
        l_vis = l_vis_list[i]
        len2 = len(l_vis['pred'])
    if r_vis_list is not None:
        r_vis = r_vis_list[i] 
        len2 = len(r_vis['pred'])
    if lr_vis_list is not None:
        lr_vis = lr_vis_list[i] 
        len2 = int(lr_vis['lidar_points'][:,0].max().item()) + 1
    TT = 0.5
    for j in range(len2):
        if lr_vis_list is not None:
            pred_scores = lr_vis['pred'][j]['pred_scores']
            pred_box = lr_vis['pred'][j]['pred_boxes'][pred_scores>TT,:]
            batch_mask = (lr_vis['lidar_points'][:,0]==j)
            r_batch_mask = (lr_vis['radar_points'][:,0]==j)
            lpoints = lr_vis['lidar_points'][batch_mask,1:4]
            rpoints = lr_vis['radar_points'][r_batch_mask,1:4]
            # if 'point_cls_scores' in lr_vis.keys():
            #     scores = lr_vis['point_cls_scores']
            #     raw_radar = lr_vis['raw_radar_points']
            #     T = 0.15
            #     rpoints = raw_radar[scores>T][:,1:4]
            # gt = gt_mask(lr_vis['gt'][j], np.concatenate([lpoints, rpoints], axis=0))
            # gt_boxes = gt[:, :-1]
            # gt_name = np.array(class_names)[np.array(gt[:, -1]).reshape(-1).astype(int)-1]
            # vi.add_points(lpoints,color='grey',radius=4)
            # vi.add_points(rpoints,scatter_filed=rpoints[:, 2],radius=8)
            # vi.add_3D_boxes(pred_box, color='blue', show_corner_spheres=False, show_heading=True, line_width=4)
            # vi.add_3D_boxes(gt_boxes, color='red', show_corner_spheres=False, show_heading=True, line_width=4, box_info=gt_name)
            # vi.show_3D()
        if l_vis_list is not None:
            pred_scores = l_vis['pred'][j]['pred_scores']
            pred_box = l_vis['pred'][j]['pred_boxes'][pred_scores>TT,:]
            batch_mask = (l_vis['points'][:,0]==j)
            lpoints = l_vis['points'][batch_mask,1:4]
            gt = gt_mask(lr_vis['gt'][j], np.concatenate([lpoints, rpoints], axis=0))
            gt_boxes = gt[:, :-1]
            gt_name = np.array(class_names)[np.array(gt[:, -1]).reshape(-1).astype(int)-1]
            vi.add_points(lpoints,color='grey',radius=4)
            vi.add_3D_boxes(pred_box, color='blue', show_corner_spheres=False, show_heading=True, line_width=4)
            vi.add_3D_boxes(gt_boxes, color='red', show_corner_spheres=False, show_heading=True, line_width=4, box_info=gt_name)
            vi.show_3D()
        # vi.add_points(rpoints,scatter_filed=rpoints[:, 2],radius=5)
        # vi.add_3D_boxes(gt_boxes, color='red', show_corner_spheres=False, show_heading=False, line_width=4)
        # vi.show_3D()
        # vi.add_points(lpoints,scatter_filed=lpoints[:, 2],radius=3)
        # vi.add_points(rpoints,color='orange',radius=8)
        # vi.add_3D_boxes(gt_boxes, color='red', show_corner_spheres=False, show_heading=False, line_width=4)
        # vi.show_3D()
```

15563264286, 2025-02-24:
好的，感谢

ylwhxht, 2025-02-24:
> 好的，感谢

我刚刚更新了一版，简单调整了一下应该会简洁一点，如果没更新请刷新一下