ylwhxht/L4DR #7: 数据集

dfxiao-134, 2025-02-21:
论文中提到了对K-Radar数据集的对比实验，想知道是怎么获取以及使用的。通过官网的服务器连接还是云盘下载

dfxiao-134, 2025-02-21:
您提到的“只能下载他们提供的稀疏4D雷达点云”这部分数据具体指的是哪一部分？不知道是否能提供具体的说明

ylwhxht, 2025-02-21:
您好，感谢您对我们工作的关注，issue中我们进行了相同的回复，有任务后续问题可以继续提出：
【4Dradar】：我们是从谷歌网盘下载的稀疏4Dradar（当时有且仅有该4D radar数据），对应的是先版本中RadarTensor的from_rdr_cube_xyz。
【LiDAR及其他标定文件】：1~58文件夹中的lpc及meta。
【标签】：我们使用的是作者建议的 'kradar_revised_label_v2_0.zip'而不是1～58中的meta文件。
【dataset版本等相关代码】：参考以下配置文件，我们也会尽快发布基于k-radar数据集实现的相关代码。
```

### ----- General ----- ###
GENERAL:
  NAME: 'PP_RLF'
  COMMENT1: '0~72, -16~16, -2~7.6'
  COMMENT2: 'CenterPoint setting'
  COMMENT3: 'Multi-class (+ Bus or Truck)'
  VERSION: '2.0'
  SEED: 2023
  IS_CUDA_SEED: True
  IS_DETERMINISTIC: True
  DEVICE: 'gpu'
  RESUME:
    IS_RESUME: False
    PATH_EXP: '/mnt/32THHD/hx/K-Radar-main/logs/PP_RLF/train_35e_0.1denoise_nofilter_nor2l_nomax' # None
    START_EP: # None
    IS_COPY_LOGS: False

  LOGGING:
    IS_LOGGING: True
    PATH_LOGGING: './logs'
    IS_SAVE_MODEL: True
    INTERVAL_EPOCH_MODEL: 1
    INTERVAL_EPOCH_UTIL: 5
### ----- General ----- ###

### ----- Dataset ----- ###
cfg_dataset_ver2: True
DATASET:
  NAME: 'KRadarDetection_v2_0'

  NUM: # Total number of frames, just make this blank

  path_data:
    list_dir_kradar: ['/mnt/32THHD/K-Radar-dataset']
    split: ['./resources/split/train.txt', './resources/split/test.txt']
    revised_label_v1_1: './tools/revise_label/kradar_revised_label_v1_1'
    revised_label_v2_0: './tools/revise_label/kradar_revised_label_v2_0/KRadar_refined_label_by_UWIPL'
    revised_label_v2_1: './tools/kradar_revised_label_v2_1/KRadar_revised_visibility'
  label_version: 'v2_1'
  item: {
    'calib': True,
    'ldr64': True,
    'ldr128': False,
    'rdr': False,
    'rdr_sparse': True,
    'cam': False,
  }
  calib: {
    'z_offset': 0.7
  }
  ldr64: {
    'processed': False,
    'skip_line': 13,
    'n_attr': 9,
    'inside_ldr64': True,
    'calib': True,
    'n_used': 4, # x, y, z, intensity
  }
  rdr: {
    'cube': False,
  }
  rdr_sparse: {
    'processed': True,
[... 258 lines cut ...]
    TYPE: 'default' # ['default', 'cube']
    DEFAULT: [0,100,-40,40,-10,60] # x_min_max, y_min_max, z_min_max / Dim: [m]

  # For BEV model visualization
  Z_CENTER: {
    'Sedan': 0.5,
    'Bus or Truck': 1.5,
    'Motorcycle': 0.5,
    'Bicycle': 0.5,
    'Pedestrian': 0.5,
    'Pedestrian Group': 0.5,
  }

  # For BEV model visualization
  Z_HEIGHT: {
    'Sedan': 1.9,
    'Bus or Truck': 1.9,
    'Motorcycle': -1,
    'Bicycle': -1,
    'Pedestrian': 2,
    'Pedestrian Group': -1,
  }
### ----- Visualization ----- ###

```

dfxiao-134, 2025-02-21:
> 您好，感谢您对我们工作的关注，issue中我们进行了相同的回复，有任务后续问题可以继续提出： 【4Dradar】：我们是从谷歌网盘下载的稀疏4Dradar（当时有且仅有该4D radar数据），对应的是先版本中RadarTensor的from_rdr_cube_xyz。 【LiDAR及其他标定文件】：1~58文件夹中的lpc及meta。 【标签】：我们使用的是作者建议的 'kradar_revised_label_v2_0.zip'而不是1～58中的meta文件。 【dataset版本等相关代码】：参考以下配置文件，我们也会尽快发布基于k-radar数据集实现的相关代码。
> 
> ```
> ### ----- General ----- ###
> GENERAL:
>   NAME: 'PP_RLF'
>   COMMENT1: 'Driving corridor, LR 0.001~0.0001'
>   COMMENT2: 'Adam'
>   VERSION: '1.1'
>   SEED: 2023
>   IS_CUDA_SEED: True
>   IS_DETERMINISTIC: True
>   DEVICE: 'gpu'
>   RESUME:
>     IS_RESUME: False
>     PATH_EXP: '/mnt/32THHD/hx/K-Radar-main/logs/PP_RLF/train240309_225904/models' # None
>     START_EP: # None
>     IS_COPY_LOGS: True
> 
>   LOGGING:
>     IS_LOGGING: True
>     PATH_LOGGING: './logs'
>     IS_SAVE_MODEL: True
>     INTERVAL_EPOCH_MODEL: 1
>     INTERVAL_EPOCH_UTIL: 5
> ### ----- General ----- ###
> 
> DATASET:
>   NAME: 'KRadarDetection_v2_0'
> 
>   NUM: # Total number of frames, just make this blank
> 
>   path_data:
>     list_dir_kradar: ['/mnt/32THHD/K-Radar-dataset']
>     split: ['./resources/split/train.txt', './resources/split/test.txt']
>     revised_label_v1_1: './tools/revise_label/kradar_revised_label_v1_1'
>     revised_label_v2_0: './tools/revise_label/kradar_revised_label_v2_0/KRadar_refined_label_by_UWIPL'
>     revised_label_v2_1: './tools/kradar_revised_label_v2_1/KRadar_revised_visibility'
>   label_version: 'v2_1'
>   item: {
>     'calib': True,
>     'ldr64': True,
>     'ldr128': False,
>     'rdr': False,
>     'rdr_sparse': True,
>     'cam': False,
>   }
>   calib: {
>     'z_offset': 0.7
>   }
>   ldr64: {
>     'processed': False,
>     'skip_line': 13,
>     'n_attr': 9,
>     'inside_ldr64': True,
>     'calib': True,
>     'n_used': 4, # x, y, z, intensity
>   }
>   rdr: {
>     'cube': False,
>   }
>   rdr_sparse: {
>     'processed': True,
>     'dir': '/mnt/32THHD/K-Radar-dataset/sparse_radar_tensor_wide_range/rtnh_wider_1p_1/',
>     'n_used': 4, # x, y, z, intensity
>     'sample_points': 10240
>   }
>   roi: {
>     'filter': True,
>     'xyz': [0.,-6.4,-2.,72.,6.4,6.0],
[... 238 lines cut ...]
>   ROI:
>     TYPE: 'default' # ['default', 'cube']
>     DEFAULT: [0,100,-40,40,-10,60] # x_min_max, y_min_max, z_min_max / Dim: [m]
> 
>   # For BEV model visualization
>   Z_CENTER: {
>     'Sedan': 0.5,
>     'Bus or Truck': 1.5,
>     'Motorcycle': 0.5,
>     'Bicycle': 0.5,
>     'Pedestrian': 0.5,
>     'Pedestrian Group': 0.5,
>   }
> 
>   # For BEV model visualization
>   Z_HEIGHT: {
>     'Sedan': 1.9,
>     'Bus or Truck': 1.9,
>     'Motorcycle': -1,
>     'Bicycle': -1,
>     'Pedestrian': 2,
>     'Pedestrian Group': -1,
>   }
> ### ----- Visualization ----- ###
> ```

dfxiao-134, 2025-02-21:
十分感谢您的回复~

closed 2025-02-21 by ylwhxht