ylwhxht/L4DR #11: 基于K-Radar数据集训练时，内存消耗问题

dfxiao-134, 2025-04-09:
根据cfg_PP_L4DR.yml训练时，内存占用一直不断增长，直至一轮epoch训练结束会重置，这个情况是使用K-Radar数据集训练普遍存在的问题吗？

ylwhxht, 2025-04-09:
> 根据cfg_PP_L4DR.yml训练时，内存占用一直不断增长，直至一轮epoch训练结束会重置，这个情况是使用K-Radar数据集训练普遍存在的问题吗？

您好，感谢您对我们L4DR工作的关注。

关于“内存一直不断增长”的程度是多少呢？以及提到的内存是否是指显卡内存？
为了保持内存不会反复变化我们在每次迭代后没有使用empty_chache清空显存，所以不会主动释放（采用torch的自主显存复用）。但是由于每次FAD去噪之后的雷达点数不同，因此一些batch的显存占用可能会变高。所以总体可能是缓慢增长。

如果并不会因此报错OOM则就是合理的，也不需要进行额外操作。如果你需要有一些其他的考虑，可以使用empty_cache清空（这样可能会导致显存变化非常频繁），但是本质上很不清空的显存峰值是一致的。

dfxiao-134, 2025-04-09:
> 您好，感谢您对我们L4DR工作的关注。
>
> 关于“内存一直不断增长”的程度是多少呢？以及提到的内存是否是指显卡内存？ 为了保持内存不会反复变化我们在每次迭代后没有使用empty_chache清空显存，所以不会主动释放（采用torch的自主显存复用）。但是由于每次FAD去噪之后的雷达点数不同，因此一些batch的显存占用可能会变高。所以总体可能是缓慢增长。 [...]

感谢您的及时回复，我的问题与显存占用无关，主要是内存(RAM)不断增长。具体来说一开始占用大约10G内存，随着一个epoch内iter增加，内存占用逐渐增大，在接近训练完一个epoch时达到最大，约80G左右。进入新的一轮epoch时内存占用会重新从10G开始，如此循环。应该是dataset预处理或者dataloader部分的原因，因为一般数据集训练内存占用应该比较固定或者变化不会很大。也可能由于K-Radar本身规模较大导致这个问题比较明显，当前这种情况导致显存没有充分利用而内存消耗过多。

ylwhxht, 2025-04-09:
> 感谢您的及时回复，我的问题与显存占用无关，主要是内存(RAM)不断增长。具体来说一开始占用大约10G内存，随着一个epoch内iter增加，内存占用逐渐增大，在接近训练完一个epoch时达到最大，约80G左右。进入新的一轮epoch时内存占用会重新从10G开始，如此循环。应该是dataset预处理或者dataloader部分的原因，因为一般数据集训练内存占用应该比较固定或者变化不会很大。也可能由于 [...]

OK，我理解了。但是我没关注内存信息，不过有注意到显卡的利用率在Kradar上并不高，并且KRadar的数据加载、evaluation感觉确实比较缓慢。

建议你可以简单训练一下最基本的PointPillar.yml来验证是否是KRadar数据集本身的问题。

但是总体上训练速度也还是勉强能够接受的，所以我没有过多去考虑这个问题。

dfxiao-134, 2025-04-09:
> OK，我理解了。但是我没关注内存信息，不过有注意到显卡的利用率在Kradar上并不高，并且KRadar的数据加载、evaluation感觉确实比较缓慢。
>
> 建议你可以简单训练一下最基本的PointPillar.yml来验证是否是KRadar数据集本身的问题。
>
> 但是总体上训练速度也还是勉强能够接受的，所以我没有过多去考虑这个问题。

感谢回复，因为之前是在大服务器上训练的，显存内存都是充分的，最近换到小的专属服务器上，因此发现了这个问题。我尝试一下原始的K-Radar项目以及您所说的PointPillar.yml。再次感谢！

ylwhxht, 2025-04-09:
> 感谢回复，因为之前是在大服务器上训练的，显存内存都是充分的，最近换到小的专属服务器上，因此发现了这个问题。我尝试一下原始的K-Radar项目以及您所说的PointPillar.yml。再次感谢！

好的，看起来非常像代码产生了内存泄露现象，可以尝试询问GPT是否有好的调试方法，或者内存管理工具例如psutil等，但是我可能本身并不是非常了解，这部分可能给不了更多建议。此外，torch.cuda.empty_cache也是用于清空显存的方法，**可能**对该问题并没有帮助。

另外推荐的一种方式是您可以计算运行时间，找到内存消耗的瓶颈模块再针对性重写/修改。

closed 2025-04-28 by ylwhxht

dfxiao-134, 2025-07-12:
您好，之前未及时回复。该数据集内存占用问题没有解决，后续训练更换了服务器，服务器在显存足够的情况下，128G内存可以满足使用。配合batchsize，numworkers参数调整可以正常训练，另外可以查看torch的dataloader内的一些参数，如prefetch_factor，pin_memory，以及persistent_workers等，看是否有帮助。

代码检查部分，初步估计是读取样本后，某些地方内存没有及时释放，导致占用不断增加，因此如果能提供的内存只是少了一部分，且只是进行整体流程测试或者可以验证方法性能，可以划分小数据集进行处理，比如每一轮随机选80%的训练样本参与训练，借助torch下的subsetrandomsampler获取train_sampler，将其传递给dataloader的sampler参数

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2025年07月07日 13:59 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [ylwhxht/L4DR] 基于K-Radar数据集训练时，内存消耗问题 (Issue #11) |
windlinsherlock left a comment (ylwhxht/L4DR#11)

根据cfg_PP_L4DR.yml训练时，内存占用一直不断增长，直至一轮epoch训练结束会重置，这个情况是使用K-Radar数据集训练普遍存在的问题吗？

您好，我在训练和测试L4DR的时候，也出现了一样的问题，请问您有解决吗？如果有解决的话，能麻烦您告诉我解决的方法吗？