ylwhxht/L4DR #12: 数据集及预训练权重请求

ttxCSDN, 2025-04-16:
您好，请问可以提供雾模拟VoD数据集及基于K-Radar数据集的预训练模型么？我的邮箱是：ttx18330267232@163.com。 我复现的结果比您论文上的指标要低几个点，想参考一下，谢谢

ttxCSDN, 2025-04-17:
非常感谢您的回复
我的复现结果主要是iou 0.3时的3d结果和iou 0.5时的 bev结果较您提供的v2.1.txt的结果低，如model34.pt的结果是：
Conf thr: 0.3, Condition: all
cls: sed
iou: 0.7 0.5 0.3 
bev: 52.59048389575326 69.30083277827225 70.69435737017261 
3d  :17.256865581414072 52.883524907584636 69.47826341026202 

Conf thr: 0.3, Condition: all
cls: bus
iou: 0.7 0.5 0.3 
bev: 28.331962071611112 53.094936019896146 63.45542060152432 
3d  :9.107184973691869 36.03280509161578 54.40108581457586 
而且我上午尝试了使用您提供的预训练权重，结果还是低于您提供的v2.1.txt。
我复现的时候测试了最后8轮的结果，我把我复现的结果和config文件以及加载您的权重的详细结果通过邮件发送给您，可以麻烦您帮忙看一下吗？十分感谢
我还有一个问题是：为什么在cfg_PP_L4DR.yml中选取radar点时只设置'sample_points': 1024，稀疏radar数据在这个roi范围下点数很多吧，只选取1024个点是否会造成信息缺失（我看cfg_PP_L4DR_v1.1.yml中是'sample_points': 10240）。

ylwhxht, 2025-04-17:
> 非常感谢您的回复 我的复现结果主要是iou 0.3时的3d结果和iou 0.5时的 bev结果较您提供的v2.1.txt的结果低，如model34.pt的结果是： Conf thr: 0.3, Condition: all cls: sed iou: 0.7 0.5 0.3 bev: 52.59048389575326 69.30083277827225 70.69435737017261 [...]

感谢您的反馈，KRadar版本的数据集和代码库都经过几次迭代版本不匹配可能会出现性能差异，我们正在重新测试提供的权重以及检查训练代码。

在此之前，您可以先进行VoD上的相关实验。

ylwhxht, 2025-04-17:
> 非常感谢您的回复 我的复现结果主要是iou 0.3时的3d结果和iou 0.5时的 bev结果较您提供的v2.1.txt的结果低，如model34.pt的结果是： Conf thr: 0.3, Condition: all cls: sed iou: 0.7 0.5 0.3 bev: 52.59048389575326 69.30083277827225 70.69435737017261 [...]

HelloHello，刚刚我重新进行了测试。似乎我搞错了配置文件，差异在于：
1. 确实应该是10240个点，时间有点久我可能对这部分细节认知有点错误。
2. 评估距离应该是更大范围的（roi的xyz是[0.,-16.,-2.,72.,16.,7.6]）之前我曾经错误上传了旧版本小范围的配置文件，后面才进行了更新，请确保你的距离是[0.,-16.,-2.,72.,16.,7.6]。

严格上说，你可以尝试使用最新版的代码+[这个权重文件](https://huggingface.co/hx24/L4DR_KRadar2.1/blob/main/model_30.pt)+以下配置文件再次进行评估，如果没问题我将会同步更新到仓库：

```
### ----- General ----- ###
GENERAL:
  NAME: 'PP_RLF'
  COMMENT1: '0~72, -16~16, -2~7.6'
  COMMENT2: 'CenterPoint setting'
  COMMENT3: 'Multi-class (+ Bus or Truck)'
  VERSION: '2.0'
  SEED: 2023
  IS_CUDA_SEED: True
  IS_DETERMINISTIC: True
  DEVICE: 'gpu'
  RESUME:
    IS_RESUME: False
    PATH_EXP: # None
    START_EP: # None
    IS_COPY_LOGS: False

  LOGGING:
    IS_LOGGING: True
    PATH_LOGGING: './logs'
    IS_SAVE_MODEL: True
    INTERVAL_EPOCH_MODEL: 1
    INTERVAL_EPOCH_UTIL: 5
### ----- General ----- ###

### ----- Dataset ----- ###
cfg_dataset_ver2: True
DATASET:
  NAME: 'KRadarDetection_v2_0'

  NUM: # Total number of frames, just make this blank

  path_data:
    list_dir_kradar: ['/mnt/32THHD/K-Radar-dataset']
    split: ['./resources/split/train.txt', './resources/split/test.txt']
    revised_label_v1_1: './tools/revise_label/kradar_revised_label_v1_1'
    revised_label_v2_0: './tools/revise_label/kradar_revised_label_v2_0/KRadar_refined_label_by_UWIPL'
    revised_label_v2_1: './tools/kradar_revised_label_v2_1/KRadar_revised_visibility'
  label_version: 'v2_1'
  item: {
    'calib': True,
    'ldr64': True,
    'ldr128': False,
    'rdr': False,
    'rdr_sparse': True,
    'cam': False,
  }
  calib: {
    'z_offset': 0.7
  }
  ldr64: {
    'processed': False,
    'skip_line': 13,
    'n_attr': 9,
    'inside_ldr64': True,
    'calib': True,
    'n_used': 4, # x, y, z, intensity
  }
  rdr: {
    'cube': False,
  }
[... 368 lines cut ...]
3d  :38.676599760610046 56.03101186538903 77.95120787488024 

Conf thr: 0.3, Condition: heavysnow
cls: sed
iou: 0.7 0.5 0.3 
bev: 46.56439035588299 53.61968471368601 60.70538038676586 
3d  :17.474873931923238 49.00583500300803 58.848390206375434 

Conf thr: 0.3, Condition: heavysnow
cls: bus
iou: 0.7 0.5 0.3 
bev: 50.614394381871854 69.41858093514638 69.52369738729098 
3d  :15.482186483012114 43.87543181401323 55.04891457240934 

Conf thr: 0.3, Condition: unnormal
cls: sed
iou: 0.7 0.5 0.3 
bev: 55.68622014273369 76.54231559019124 79.2439044791892 
3d  :17.23215609639913 54.72574575543052 76.46394123448242 

Conf thr: 0.3, Condition: unnormal
cls: bus
iou: 0.7 0.5 0.3 
bev: 45.98044106867082 63.752677175521924 71.14469487913856 
3d  :19.591579564865388 46.91848119091897 59.3102620587458

ttxCSDN, 2025-04-17:
收到，非常感谢您的回复和测试。
我再去测试一下，有结果之后再和您联系。谢谢您！

---- 回复的原邮件 ----
| 发件人 | Xun ***@***.***> |
| 日期 | 2025年04月17日 15:50 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [ylwhxht/L4DR] 数据集及预训练权重请求 (Issue #12) |

非常感谢您的回复 我的复现结果主要是iou 0.3时的3d结果和iou 0.5时的 bev结果较您提供的v2.1.txt的结果低，如model34.pt的结果是： Conf thr: 0.3, Condition: all cls: sed iou: 0.7 0.5 0.3 bev: 52.59048389575326 69.30083277827225 70.69435737017261 3d :17.256865581414072 52.883524907584636 69.47826341026202

Conf thr: 0.3, Condition: all cls: bus iou: 0.7 0.5 0.3 bev: 28.331962071611112 53.094936019896146 63.45542060152432 3d :9.107184973691869 36.03280509161578 54.40108581457586 而且我上午尝试了使用您提供的预训练权重，结果还是低于您提供的v2.1.txt。 我复现的时候测试了最后8轮的结果，我把我复现的结果和config文件以及加载您的权重的详细结果通过邮件发送给您，可以麻烦您帮忙看一下吗？十分感谢 我还有一个问题是：为什么在cfg_PP_L4DR.yml中选取radar点时只设置'sample_points': 1024，稀疏radar数据在这个roi范围下点数很多吧，只选取1024个点是否会造成信息缺失（我看cfg_PP_L4DR_v1.1.yml中是'sample_points': 10240）。

HelloHello，刚刚我重新进行了测试。似乎我搞错了配置文件，差异在于：

确实应该是10240个点，时间有点久我可能对这部分细节认知有点错误。
评估距离应该是更大范围的（roi的xyz是[0.,-16.,-2.,72.,16.,7.6]）之前我曾经错误上传了旧版本小范围的配置文件，后面才进行了更新，请确保你的距离是[0.,-16.,-2.,72.,16.,7.6]。

严格上说，你可以尝试使用最新版的代码+这个权重文件+以下配置文件再次进行评估，如果没问题我将会同步更新到仓库：

### ----- General ----- ###
GENERAL:
  NAME: 'PP_RLF'
  COMMENT1: '0~72, -16~16, -2~7.6'
  COMMENT2: 'CenterPoint setting'
  COMMENT3: 'Multi-class (+ Bus or Truck)'
  VERSION: '2.0'
  SEED: 2023
  IS_CUDA_SEED: True
  IS_DETERMINISTIC: True
  DEVICE: 'gpu'
  RESUME:
    IS_RESUME: False
    PATH_EXP: # None
    START_EP: # None
    IS_COPY_LOGS: False

  LOGGING:
    IS_LOGGING: True
    PATH_LOGGING: './logs'
    IS_SAVE_MODEL: True
    INTERVAL_EPOCH_MODEL: 1
    INTERVAL_EPOCH_UTIL: 5
### ----- General ----- ###

### ----- Dataset ----- ###
cfg_dataset_ver2: True
DATASET:
  NAME: 'KRadarDetection_v2_0'

  NUM: # Total number of frames, just make this blank

  path_data:
    list_dir_kradar: ['/mnt/32THHD/K-Radar-dataset']
    split: ['./resources/split/train.txt', './resources/split/test.txt']
    revised_label_v1_1: './tools/revise_label/kradar_revised_label_v1_1'
    revised_label_v2_0: './tools/revise_label/kradar_revised_label_v2_0/KRadar_refined_label_by_UWIPL'
    revised_label_v2_1: './tools/kradar_revised_label_v2_1/KRadar_revised_visibility'
  label_version: 'v2_1'
  item: {
    'calib': True,
    'ldr64': True,
    'ldr128': False,
    'rdr': False,
    'rdr_sparse': True,
    'cam': False,
  }
  calib: {
    'z_offset': 0.7
[... 844 lines cut ...]
3d :38.676599760610046 56.03101186538903 77.95120787488024

Conf thr: 0.3, Condition: heavysnow
cls: sed
iou: 0.7 0.5 0.3
bev: 46.56439035588299 53.61968471368601 60.70538038676586
3d :17.474873931923238 49.00583500300803 58.848390206375434

Conf thr: 0.3, Condition: heavysnow
cls: bus
iou: 0.7 0.5 0.3
bev: 50.614394381871854 69.41858093514638 69.52369738729098
3d :15.482186483012114 43.87543181401323 55.04891457240934

Conf thr: 0.3, Condition: unnormal
cls: sed
iou: 0.7 0.5 0.3
bev: 55.68622014273369 76.54231559019124 79.2439044791892
3d :17.23215609639913 54.72574575543052 76.46394123448242

Conf thr: 0.3, Condition: unnormal
cls: bus
iou: 0.7 0.5 0.3
bev: 45.98044106867082 63.752677175521924 71.14469487913856
3d :19.591579564865388 46.91848119091897 59.3102620587458

ttxCSDN, 2025-04-20:
您好，我的训练和测试都有了结果，但还是有些不一致。
我使用您提供的配置文件+权重进行测试，得到的结果是：（与您提供的结果基本一致）
Conf thr: 0.3, Condition: all
cls: sed
iou: 0.7 0.5 0.3 
bev: 50.962600940862835 75.2853691587103 77.70246381286586 
3d  :11.52968964369597 51.58152975068388 75.81355508038055 

Conf thr: 0.3, Condition: all
cls: bus
iou: 0.7 0.5 0.3 
bev: 32.19870940572581 58.28537098425035 64.27601475586367 
3d  :10.418014773353613 43.35160154854003 59.36793286560808 
但使用您提供的配置文件训练35轮，使用最好的一轮来测试，结果也只是：（是不是和环境以及服务器还是有很大影响，这个训练出来的差别还是有点大）
Conf thr: 0.3, Condition: all
cls: sed
iou: 0.7 0.5 0.3 
bev: 50.95128967399049 68.9903642043006 77.81122679018961 
3d  :12.833549178621917 51.29537369524348 69.21382724651174 

Conf thr: 0.3, Condition: all
cls: bus
iou: 0.7 0.5 0.3 
bev: 26.834895535973867 52.55809291398376 62.79702746973128 
3d  :9.09215868179824 33.4273426643293 55.576438117222715 

Conf thr: 0.3, Condition: normal
cls: sed
iou: 0.7 0.5 0.3 
bev: 42.563051445910496 67.65850894209072 76.47681349679563 
3d  :8.662640907510198 48.75991936138159 68.10556803296957 

Conf thr: 0.3, Condition: normal
cls: bus
iou: 0.7 0.5 0.3 
bev: 20.40704001208281 50.5794269063228 62.46713614902718 
3d  :7.184984493828744 33.452058488935585 55.79089807740408 

Conf thr: 0.3, Condition: overcast
cls: sed
iou: 0.7 0.5 0.3 
bev: 49.511319280754016 78.99192214439249 88.9399645764158 
3d  :15.103052479243257 52.363233435853076 79.2440050922993 

Conf thr: 0.3, Condition: overcast
cls: bus
iou: 0.7 0.5 0.3 
bev: 33.34721542258304 77.1573290573612 80.23349704958312 
3d  :10.803077211745943 59.75897836359979 80.23349704958312 

Conf thr: 0.3, Condition: fog
cls: sed
iou: 0.7 0.5 0.3 
bev: 77.7402927742046 90.27917534886014 90.36011671228643 
3d  :42.8496653696641 77.08554707917436 88.64320656260382 

Conf thr: 0.3, Condition: fog
cls: bus
iou: 0.7 0.5 0.3 
bev: 0.0 0.0 0.0 
3d  :0.0 0.0 0.0 

Conf thr: 0.3, Condition: rain
cls: sed
iou: 0.7 0.5 0.3 
bev: 52.24347765584885 77.45858204275687 79.64745585535209 
3d  :13.181492638377664 51.06851613856151 70.69690000593343 

Conf thr: 0.3, Condition: rain
cls: bus
iou: 0.7 0.5 0.3 
bev: 2.6607538802660753 2.6936026936026933 3.325942350332594 
3d  :0.11086474501108648 2.6607538802660753 2.767646427805192 

Conf thr: 0.3, Condition: sleet
cls: sed
iou: 0.7 0.5 0.3 
bev: 39.05098275510781 51.28482964959058 60.48715394108537 
3d  :17.220584546198854 39.29958790471606 57.428876377390836 

Conf thr: 0.3, Condition: sleet
cls: bus
iou: 0.7 0.5 0.3 
bev: 25.80967496297914 45.121271258763315 56.96720682412296 
3d  :7.353434839770266 38.80318379362375 56.00931640908221 

Conf thr: 0.3, Condition: lightsnow
cls: sed
iou: 0.7 0.5 0.3 
bev: 63.68679041640433 78.0078178656899 85.361923257538 
3d  :16.801548979509473 60.29167584817151 76.89587652941398 

Conf thr: 0.3, Condition: lightsnow
cls: bus
iou: 0.7 0.5 0.3 
bev: 59.02518549569717 72.51532535591106 81.86472413281088 
3d  :36.065632786168344 58.843209996266054 77.40942884354784 

Conf thr: 0.3, Condition: heavysnow
cls: sed
iou: 0.7 0.5 0.3 
bev: 41.41793586317308 52.99757270585243 53.587582222160805 
3d  :19.67056379275978 40.45572336373605 52.724026004201285 

Conf thr: 0.3, Condition: heavysnow
cls: bus
iou: 0.7 0.5 0.3 
bev: 46.61550939032067 69.82165381516032 70.02596400056156 
3d  :11.645877370999075 24.702745016991845 42.95633789613466 

Conf thr: 0.3, Condition: unnormal
cls: sed
iou: 0.7 0.5 0.3 
bev: 54.470897683344354 70.32369222057065 78.94643990167513 
3d  :20.291130137476497 53.72191828162764 70.28963433449414 

Conf thr: 0.3, Condition: unnormal
cls: bus
iou: 0.7 0.5 0.3 
bev: 36.15940682954793 61.19303896118811 64.04786331110385 
3d  :13.470946838250144 33.389935802309886 52.78854327901933

ylwhxht, 2025-04-20:
> 您好，我的训练和测试都有了结果，但还是有些不一致。 我使用您提供的配置文件+权重进行测试，得到的结果是：（与您提供的结果基本一致） Conf thr: 0.3, Condition: all cls: sed iou: 0.7 0.5 0.3 bev: 50.962600940862835 75.2853691587103 77.70246381286586 3d [...]

> 您好，我的训练和测试都有了结果，但还是有些不一致。 我使用您提供的配置文件+权重进行测试，得到的结果是：（与您提供的结果基本一致） Conf thr: 0.3, Condition: all cls: sed iou: 0.7 0.5 0.3 bev: 50.962600940862835 75.2853691587103 77.70246381286586 3d [...]

看起来在Sed的BEV_0.5、3D_0.5；Bus的大部分指标上性能有较大差异。在我们前期实验的时候就发现K-Radar上的性能波动非常大，建议可以重新训练以及换为30epoch进行尝试看看变化。

我们将尽可能排查一下问题，但是由于K-Radar的数据、标签、代码库均发生了调整，并且这部分工作也是比较久之前完成的了，是否能完整复原K-Radar训练过程可能存疑。但是因为VoD的数据版本未变且评估性能波动较小，我们在VoD上的结果应该相对稳定，有人也成功复现了。
综上，在我们排查的时间里，建议保留一些备选研究方案：
1. 尝试复现v1.0结果、调整epoch等参数重新训练。
2. 基于我们VoD-fog上的系列性能作为主要表格与汇报精度进行对比，K-Radar都按照官方提供的最新数据、最新标签、最新代码库复现，以自己复现精度进行对比。
3. 如果实在无法解决，可能要考虑调整baseline：K-Radar作者最近发布了一个新工作[ASF](https://arxiv.org/abs/2503.07029)，论文所性能显著高于我们，可能还需要关注一下他们的工作作为baseline比较助于性能的提升。

ylwhxht, 2025-04-20:
> 看起来在Sed的BEV_0.5、3D_0.5；Bus的大部分指标上性能有较大差异。在我们前期实验的时候就发现K-Radar上的性能波动非常大，建议可以重新训练以及换为30epoch进行尝试看看变化。
>
> 我们将尽可能排查一下问题，但是由于K-Radar的数据、标签、代码库均发生了调整，并且这部分工作也是比较久之前完成的了，是否能完整复原K-Radar训练过程可能存疑。

因为VoD的数据版本未变且评估性能波动较小，我们在VoD上的结果应该相对稳定，有人也成功复现了。 综上，在我们排查的时间里，建议保留一些备选研究方案：
> 1. 尝试复现v1.0结果、调整epoch等参数重新训练。
> 2. 基于我们VoD-fog上的系列性能作为主要表格与汇报精度进行对比，K-Radar都按照官方提供的最新数据、最新标签、最新代码库复现，以自己复现精度进行对比。
> 3. 如果实在无法解决，可能要考虑调整baseline：K-Radar作者最近发布了一个新工作[ASF](https://arxiv.org/abs/2503.07029)，论文所性能显著高于我们，可能还需要关注一下他们的工作作为baseline比较助于性能的提升。

ttxCSDN, 2025-04-20:
感谢您的回复，我会再次尝试进行v1.0版本的复现以及参数的调整

ylwhxht, 2025-04-24:
> 感谢您的回复，我会再次尝试进行v1.0版本的复现以及参数的调整

您好，我们尝试重新下载数据、标签以及代码配置之后确实难以复现出相当的结果，但是评估的时候通过调整Confidence为0.1可以得到近似的结果。（注：原论文汇报的结果是在0.3对齐的Confidence下实现的，但是Confidence在目标检测中并非不可调整的超参数）所以你可以尝试评估Confidence为0.1下的性能。

我们也将尽力对比两个版本的代码定位问题尝试恢复Confidence=0.3下复现性能。

closed 2025-04-24 by ylwhxht