ylwhxht/L4DR #14: 作者你好，能求一份VOD数据集的预训练模型吗，自己训练完离论文还有差距

czxv587, 2025-04-24:


czxv587, 2025-04-24:
邮箱2205850061@qq.com

ylwhxht, 2025-04-24:
> 邮箱[2205850061@qq.com](mailto:2205850061@qq.com)

此外你也可以提供一下你训练的参数，我们推荐使用：batchsize=8，epoch=100

czxv587, 2025-04-24:
> > 邮箱[2205850061@qq.com](mailto:2205850061@qq.com)
> 
> 此外你也可以提供一下你训练的参数，我们推荐使用：batchsize=8，epoch=100

好的，谢谢作者了，之前训练了是用的batchsize=8，epoch=100，但我看没拟合就又加了20轮，现在是120轮，只做了这一个修改，然后使用最后一轮的模型测试的（不知道是不是最好的），然后刚刚用预训练模型测试了一下，高了不少
我测试的：
2025-04-24 16:00:27,033   INFO  *************** EPOCH 120 EVALUATION *****************
2025-04-24 16:04:08,247   INFO  *************** Performance of EPOCH 120 *****************
2025-04-24 16:04:08,247   INFO  Generate label finished(sec_per_example: 0.1707 second).
2025-04-24 16:04:08,247   INFO  recall_roi_0.3: 0.000000
2025-04-24 16:04:08,247   INFO  recall_rcnn_0.3: 0.752903
2025-04-24 16:04:08,247   INFO  recall_roi_0.5: 0.000000
2025-04-24 16:04:08,247   INFO  recall_rcnn_0.5: 0.659172
2025-04-24 16:04:08,247   INFO  recall_roi_0.7: 0.000000
2025-04-24 16:04:08,247   INFO  recall_rcnn_0.7: 0.319400
2025-04-24 16:04:08,248   INFO  Average predicted number of objects(1296 samples): 10.327
2025-04-24 16:04:19,773   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:64.5603, 66.1791, 60.0432
bev  AP:70.7013, 67.5680, 60.8503
3d   AP:50.5988, 48.0748, 41.9372
aos  AP:61.28, 64.19, 58.30
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:65.7814, 65.2970, 60.1537
bev  AP:70.7727, 67.7102, 60.9980
3d   AP:49.5706, 46.1428, 41.3179
aos  AP:62.07, 63.22, 58.21
Car AP@0.70, 0.50, 0.50:
bbox AP:64.5603, 66.1791, 60.0432
bev  AP:82.8913, 78.5369, 70.9799
3d   AP:74.0763, 70.4438, 67.6077
aos  AP:61.28, 64.19, 58.30
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:65.7814, 65.2970, 60.1537
bev  AP:84.4633, 79.2954, 74.1076
3d   AP:75.1964, 72.9979, 66.4243
aos  AP:62.07, 63.22, 58.21
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:70.5511, 66.1591, 62.9091
bev  AP:64.6668, 61.9575, 55.9180
3d   AP:59.3046, 54.3434, 47.8169
aos  AP:68.22, 62.74, 59.27
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:70.2240, 66.8206, 61.6413
bev  AP:65.8579, 60.9097, 55.6043
3d   AP:58.2834, 53.7199, 47.6015
aos  AP:67.82, 63.05, 57.83
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:70.5511, 66.1591, 62.9091
bev  AP:72.5190, 67.3793, 64.5183
3d   AP:71.9427, 67.1238, 64.1083
aos  AP:68.22, 62.74, 59.27
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:70.2240, 66.8206, 61.6413
bev  AP:73.1654, 69.4137, 64.3236
3d   AP:72.8211, 69.0809, 64.0095
aos  AP:67.82, 63.05, 57.83
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:87.6129, 86.1073, 78.3626
bev  AP:87.8435, 86.2965, 78.4508
3d   AP:87.3459, 83.6274, 76.7076
aos  AP:87.34, 85.80, 78.06
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:92.1666, 87.2805, 79.9114
bev  AP:92.4879, 87.5004, 79.8877
3d   AP:89.7719, 84.8749, 77.2995
aos  AP:91.93, 86.99, 79.60
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:87.6129, 86.1073, 78.3626
bev  AP:87.9570, 87.2175, 78.8920
3d   AP:87.9570, 87.1708, 78.8855
aos  AP:87.34, 85.80, 78.06
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:92.1666, 87.2805, 79.9114
bev  AP:92.7024, 87.9275, 82.1751
3d   AP:92.6947, 87.9074, 81.9815
aos  AP:91.93, 86.99, 79.60
预训练模型：
2025-04-24 16:15:51,039   INFO  *************** EPOCH 4 EVALUATION *****************
2025-04-24 16:18:21,352   INFO  *************** Performance of EPOCH 4 *****************
2025-04-24 16:18:21,352   INFO  Generate label finished(sec_per_example: 0.1160 second).
2025-04-24 16:18:21,352   INFO  recall_roi_0.3: 0.000000
2025-04-24 16:18:21,352   INFO  recall_rcnn_0.3: 0.776863
2025-04-24 16:18:21,352   INFO  recall_roi_0.5: 0.000000
2025-04-24 16:18:21,352   INFO  recall_rcnn_0.5: 0.677327
2025-04-24 16:18:21,352   INFO  recall_roi_0.7: 0.000000
2025-04-24 16:18:21,352   INFO  recall_rcnn_0.7: 0.343044
2025-04-24 16:18:21,352   INFO  Average predicted number of objects(1296 samples): 11.837
2025-04-24 16:18:31,373   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:72.0714, 69.5864, 61.5052
bev  AP:69.7952, 67.4841, 60.6071
3d   AP:53.2569, 50.8227, 46.2290
aos  AP:66.22, 66.69, 58.98
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:72.8413, 70.3744, 63.3719
bev  AP:70.5662, 67.3758, 60.6287
3d   AP:52.9862, 49.4465, 44.2238
aos  AP:66.27, 67.13, 60.38
Car AP@0.70, 0.50, 0.50:
bbox AP:72.0714, 69.5864, 61.5052
bev  AP:86.1153, 79.5629, 76.5741
3d   AP:81.3257, 76.2570, 69.9493
aos  AP:66.22, 66.69, 58.98
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:72.8413, 70.3744, 63.3719
bev  AP:88.3546, 81.4429, 75.9884
3d   AP:83.0870, 76.5604, 69.4318
aos  AP:66.27, 67.13, 60.38
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:72.3296, 70.8985, 64.4136
bev  AP:64.8103, 62.4631, 56.1154
3d   AP:58.7787, 54.9643, 48.2272
aos  AP:69.37, 66.14, 59.86
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:73.4453, 70.6430, 64.0664
bev  AP:65.9623, 62.0494, 55.4683
3d   AP:58.2298, 54.1037, 47.8127
aos  AP:70.17, 65.48, 59.03
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:72.3296, 70.8985, 64.4136
bev  AP:73.5398, 71.9305, 65.5412
3d   AP:73.4463, 71.6948, 65.2834
aos  AP:69.37, 66.14, 59.86
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:73.4453, 70.6430, 64.0664
bev  AP:74.5844, 72.6442, 66.0944
3d   AP:74.4895, 72.4494, 65.8758
aos  AP:70.17, 65.48, 59.03
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:88.2207, 87.6956, 79.0784
bev  AP:88.1205, 87.3548, 78.7999
3d   AP:87.1818, 84.7457, 77.8739
aos  AP:88.01, 87.47, 78.87
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:93.2093, 90.0153, 82.7001
bev  AP:93.0231, 89.4137, 80.5842
3d   AP:91.8778, 86.7192, 77.7248
aos  AP:92.99, 89.76, 82.43
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:88.2207, 87.6956, 79.0784
bev  AP:88.2745, 87.8566, 79.1986
3d   AP:88.2745, 87.8566, 79.1986
aos  AP:88.01, 87.47, 78.87
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:93.2093, 90.0153, 82.7001
bev  AP:93.3023, 90.5515, 83.1662
3d   AP:93.2987, 90.5428, 83.1074
aos  AP:92.99, 89.76, 82.43

ylwhxht, 2025-04-24:
> 好的，谢谢作者了，之前训练了是用的batchsize=8，epoch=100，但我看没拟合就又加了20轮，现在是120轮，只做了这一个修改，然后使用最后一轮的模型测试的（不知道是不是最好的），然后刚刚用预训练模型测试了一下，高了不少 我测试的： 2025-04-24 16:00:27,033 INFO *************** EPOCH 120 EVALUATION [...]

一些问题：
1. 训练100轮的结果是否有进行过测试？
2. “就又加了20轮”是重新开始训练的吗？
3. 最后一轮一般不是最好的，我们是选取最好的一个epoch，但是性能之间也并不会差距太大。

czxv587, 2025-04-24:
> 一些问题：
>
> 1. 训练100轮的结果是否有进行过测试？
> 2. “就又加了20轮”是重新开始训练的吗？
> 3. 最后一轮一般不是最好的，我们是选取最好的一个epoch，但是性能之间也并不会差距太大。

第100轮的结果测试过和第120轮差别不大，加了20轮是在100的断点处继续训练的

ylwhxht, 2025-04-24:
> batchsize=8，epoch=100

> 第100轮的结果测试过和第120轮差别不大，加了20轮是在100的断点处继续训练的

不能续加断点，训练使用的学习率等等都是随着epoch固定的，如果调整了epoch只能从头训练，如果你100个的结果并不理想，建议重头训120个尝试一下。

虽然我们在训练的时候只训练了100个epoch，可能是由于gpu数导致的总batchsize不同。

czxv587, 2025-04-24:
> > 批量大小 = 8，纪元 = 100
> 
> > > > > > 邮箱 [2205850061@qq.com](mailto:2205850061@qq.com)
> > > > > 
> > > > > 
> > > > > 此外你也可以提供一下你训练的参数，我们推荐使用：batchsize=8，epoch=100
> > > > 
> > > > 
> > > > 好的，谢谢作者了，之前训练了是用的batchsize=8，epoch=100，但我看没拟合就又加了20轮，现在是120轮，只做了这一个修改，然后使用最后一轮的模型测试的（不知道是不是最好的），然后刚刚用预训练模型测试了一下，高了不少 我测试的： 2025-04-24 16：00：27,033 INFO *************** EPOCH 120评估***************** 2025-04-24 16：04：08,247 INFO *************** EPOCH 120的性能 ***************** 2025-04-24 16：04：08,247 INFO 生成标签完成（sec_per_example：0.1707秒）。2025-04-24 16：04：08,247 信息 recall_roi_0.3：0.000000 2025-04-24 16：04：08,247 信息 recall_rcnn_0.3：0.752903 2025-04-24 16：04：08,247 信息 recall_roi_0.5： 0.000000 2025-04-24 16：04：08,247 信息 recall_rcnn_0.5： 0.659172 2025-04-24 16：04：08,247 信息 recall_roi_0.7： 0.000000 2025-04-24 16：04：08,247 信息 recall_rcnn_0.7： 0.319400 2025-04-24 16：04：08,248 信息 平均预测对象数（1296个样本）：10.327 2025-04-24 16：04：19,773 信息 汽车 AP@0.70， 0.70， 0.70： bbox AP：64.5603， 66.1791， 60.0432 bev AP：70.7013， 67. 5680， 60.8503 3d AP：50.5988， 48.0748， 41.9372 aos AP：61.28， 64.19， 58.30 汽车 AP_R40@0.70， 0.70， 0.70： bbox AP：65.7814， 65.2970， 60.1537 bev AP：70.7727， 67.7102， 60.9980 3d AP：49.5706， 46.1428， 41.3179 aos AP：62.07， 63.22， 58.21 汽车 AP@0.70， 0.50， 0.50： bbox AP：64.5603， 66.1791， 60.0432 bev AP：82.8913， 78.5369， 70.9799 3d AP：74.0763， 70.4438， 67.6077 aos AP：61.28， 64. 19， 58.30 汽车 AP_R40@0.70， 0.50， 0.50： bbox AP：65.7814， 65.2970， 60.1537 bev AP：84.4633， 79.2954， 74.1076 3d AP：75.1964， 72.9979， 66.4243 aos AP：62.07， 63.22， 58.21 行人 AP@0.50， 0.50， 0.50： bbox AP：70.5511， 66.1591， 62.9091 bev AP：64.6668， 61.9575， 55.9180 3d AP：59.3046， 54.3434， 47.8169 aos AP：68.22， 62.74、59.27 行人 AP_R40@0.50、0.50、0.50：bbox AP： 70.2240， 66.8206， 61.6413 bev AP：65.8579， 60.9097， 55.6043 3d AP：58.2834， 53.7199， 47.6015 aos AP：67.82， 63.05， 57.83 行人AP@0.50， 0.25， 0.25： bbox AP：70.5511， 66.1591， 62.9091 bev AP：72.5190， 67.3793， 64.5183 3d AP：71.9427， 67.1238， 64.1083 aos AP：68.22， 62.74， 59.27 行人AP_R40@0.50， 0.25， 0.25： bbox AP：70.2240， 66.8206， 61.6413 bev AP：73.1654， 69. 4137， 64.3236 3d AP：72.8211， 69.0809， 64.0095 aos AP：67.82， 63.05， 57.83 骑行者 AP@0.50， 0.50， 0.50： bbox AP：87.6129， 86.1073， 78.3626 bev AP：87.8435， 86.2965， 78.4508 3d AP：87.3459， 83.6274， 76.7076 aos AP：87.34， 85.80， 78.06 骑行者 AP_R40@0.50， 0.50， 0.50： bbox AP：92.1666， 87.2805， 79.9114 bev AP：92.4879， 87.5004， 79.8877 3d AP：89.7719， 84.8749， 77.2995 aos AP：91.93， 86. 99， 79.60 骑行者 AP@0.50， 0.25， 0.25： bbox AP：87.6129， 86.1073， 78.3626 bev AP：87.9570， 87.2175， 78.8920 3d AP：87.9570， 87.1708， 78.8855 aos AP：87.34， 85.80， 78.06 骑行者 AP_R40@0.50， 0.25， 0.25： bbox AP：92.1666， 87.2805， 79.9114 bev AP：92.7024， 87.9275， 82.1751 3d AP：92.6947， 87.9074， 81.9815 aos AP：91.93， 86.99， 79.60 预训练模型： 2025-04-24 16：15：51,039 INFO *************** EPOCH 4 评估 ***************** 2025-04-24 16：18：21,352 INFO *************** EPOCH 4 的性能 ***************** 2025-04-24 16：18：21,352 INFO 生成标签完成（sec_per_example：0.1160 秒）。2025-04-24 16：18：21,352 信息 recall_roi_0.3：0.000000 2025-04-24 16：18：21,352 信息 recall_rcnn_0.3：0.776863 2025-04-24 16：18：21,352 信息 recall_roi_0.5： 0.000000 2025-04-24 16：18：21,352 信息 recall_rcnn_0.5： 0.677327 2025-04-24 16：18：21,352 信息 recall_roi_0.7： 0.000000 2025-04-24 16：18：21,352 信息 recall_rcnn_0.7： 0.343044 2025-04-24 16：18：21,352 信息 平均预测对象数（1296个样本）：11.837 2025-04-24 16：18：31,373 信息 汽车 AP@0.70， 0.70， 0.70： bbox AP：72.0714， 69.5864， 61.5052 bev AP：69.7952， 67.4841， 60.60713d AP：53.2569， 50.8227， 46.2290 aos AP：66. 22， 66.69， 58.98 汽车 AP_R40@0.70， 0.70， 0.70： bbox AP：72.8413， 70.3744， 63.3719 bev AP：70.5662， 67.3758， 60.6287 3d AP：52.9862， 49.4465， 44.2238 aos AP：66.27， 67.13， 60.38 汽车 AP@0.70， 0.50， 0.50： bbox AP：72.0714， 69.5864， 61.5052 bev AP：86.1153， 79.5629， 76.5741 3d AP：81.3257， 76.2570， 69.9493 aos AP：66.22， 66.69、58.98 汽车 AP_R40@0.70、0.50、0.50：bbox AP： 72.8413， 70.3744， 63.3719 bev AP：88.3546， 81.4429， 75.9884 3d AP：83.0870， 76.5604， 69.4318 aos AP：66.27， 67.13， 60.38 行人 AP@0.50， 0.50， 0.50： bbox AP：72.3296， 70.8985， 64.4136 bev AP：64.8103， 62.4631， 56.1154 3d AP：58.7787， 54.9643， 48.2272 aos AP：69.37， 66.14， 59.86 行人AP_R40@0.50， 0.50， 0.50： bbox AP：73.4453， 70.6430， 64.0664 bev AP：65.9623， 62. 0494， 55.4683 3d AP：58.2298， 54.1037， 47.8127 aos AP：70.17， 65.48， 59.03 行人AP@0.50， 0.25， 0.25： bbox AP：72.3296， 70.8985， 64.4136 bev AP：73.5398， 71.9305， 65.5412 3d AP：73.4463， 71.6948， 65.2834 aos AP：69.37， 66.14， 59.86 行人AP_R40@0.50， 0.25， 0.25： bbox AP：73.4453， 70.6430， 64.0664 bev AP：74.5844， 72.6442， 66.0944 3d AP：74.4895， 72.4494， 65.8758 aos AP：70.17， 65. 48， 59.03 Cyclist AP@0.50， 0.50， 0.50： bbox AP：88.2207， 87.6956， 79.0784 bev AP：88.1205， 87.3548， 78.7999 3d AP：87.1818， 84.7457， 77.8739 aos AP：88.01， 87.47， 78.87 Cyclist AP_R40@0.50， 0.50， 0.50： bbox AP：93.2093， 90.0153， 82.7001 bev AP：93.0231， 89.4137， 80.5842 3d AP：91.8778， 86.7192， 77.7248 aos AP：92.99， 89.76， 82.43 骑自行车者 AP@0.. 50， 0.25， 0.25： bbox AP：88.2207， 87.6956， 79.0784 bev AP：88.2745， 87.8566， 79.1986 3d AP：88.2745， 87.8566， 79.1986 aos AP：88.01， 87.47， 78.87 Cyclist AP_R40@0.50， 0.25， 0.25： bbox AP：93.2093， 90.0153， 82.7001 bev AP：93.3023， 90.5515， 83.1662 3d AP：93.2987， 90.5428， 83.1074 aos AP：92.99， 89.76， 82.43
> > > 
> > > 
> > > 一些问题：
> > > 
> > > 1. 训练100轮的结果是否有进行过测试？
> > > 2. “就又加了20轮”是重新开始训练的吗？
> > > 3. 最后一轮一般不是最好的，我们是选取最好的一个epoch，但是性能之间也并不会差距太大。
> > 
> > 
> > 第100轮的结果测试过和第120轮差别不大，加了20轮是在100的断点处继续训练的
> 
> 不能续加断点，训练使用的学习率等等都是随着epoch固定的，如果调整了epoch只能从头训练，如果你100个的结果并不理想，建议重头训120个尝试一下。
> 
> 虽然我们在训练的时候只训练了100个epoch，可能是由于gpu数导致的总batchsize不同。

好的谢谢作者解答，我再训练看看

ylwhxht, 2025-04-24:
> 好的谢谢作者解答，我再训练看看

好的，另外如果你使用了多卡训练，请加上--sync_bn

czxv587, 2025-04-27:
> 好的，另外如果你使用了多卡训练，请加上--sync_bn

你好作者，我新训练了120轮，单看其中car 3d检测精度，第97轮最高，为81.7157，预训练模型是83.0870，论文是85，还是差了些，我也不知道问题出在哪儿
下面是第97轮的测试log
INFO  *************** EPOCH 97 EVALUATION *****************
2025-04-27 13:54:30,580   INFO  *************** Performance of EPOCH 97 *****************
2025-04-27 13:54:30,580   INFO  Generate label finished(sec_per_example: 0.0524 second).
2025-04-27 13:54:30,580   INFO  recall_roi_0.3: 0.000000
2025-04-27 13:54:30,580   INFO  recall_rcnn_0.3: 0.767680
2025-04-27 13:54:30,580   INFO  recall_roi_0.5: 0.000000
2025-04-27 13:54:30,580   INFO  recall_rcnn_0.5: 0.667933
2025-04-27 13:54:30,580   INFO  recall_roi_0.7: 0.000000
2025-04-27 13:54:30,580   INFO  recall_rcnn_0.7: 0.338083
2025-04-27 13:54:30,581   INFO  Average predicted number of objects(1296 samples): 10.586
2025-04-27 13:54:33,063   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:70.1269, 69.7976, 61.7146
bev  AP:70.3594, 68.6892, 61.2688
3d   AP:50.6262, 50.3377, 46.6318
aos  AP:63.50, 66.87, 59.19
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:70.4151, 69.2228, 63.7739
bev  AP:71.4150, 68.4731, 61.4514
3d   AP:48.3007, 49.1253, 43.9052
aos  AP:63.29, 66.12, 60.84
Car AP@0.70, 0.50, 0.50:
bbox AP:70.1269, 69.7976, 61.7146
bev  AP:86.7699, 80.0790, 71.8426
3d   AP:80.8536, 71.6879, 69.2530
aos  AP:63.50, 66.87, 59.19
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:70.4151, 69.2228, 63.7739
bev  AP:89.0870, 82.3368, 75.2342
3d   AP:81.7157, 74.9689, 69.4639
aos  AP:63.29, 66.12, 60.84
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:72.4369, 66.8949, 63.7340
bev  AP:69.2615, 62.9744, 56.5908
3d   AP:59.8310, 54.6700, 48.1846
aos  AP:69.53, 62.99, 59.49
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:73.2203, 68.7372, 63.4673
bev  AP:68.8802, 62.6996, 56.1524
3d   AP:59.1461, 54.0643, 47.8334
aos  AP:70.17, 64.19, 58.79
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:72.4369, 66.8949, 63.7340
bev  AP:74.2048, 71.4547, 65.2400
3d   AP:74.1340, 71.3967, 64.9572
aos  AP:69.53, 62.99, 59.49
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:73.2203, 68.7372, 63.4673
bev  AP:75.7705, 71.1551, 65.8357
3d   AP:75.6909, 71.0701, 64.6195
aos  AP:70.17, 64.19, 58.79
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:87.7211, 86.9261, 78.6106
bev  AP:87.6988, 86.8658, 78.5801
3d   AP:87.0770, 84.9890, 77.5743
aos  AP:87.57, 86.74, 78.44
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:92.9081, 88.0743, 80.6394
bev  AP:92.8544, 88.0201, 80.5841
3d   AP:92.0917, 86.9347, 77.8515
aos  AP:92.73, 87.86, 80.40
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:87.7211, 86.9261, 78.6106
bev  AP:87.7550, 87.2093, 78.7211
3d   AP:87.7550, 87.2093, 78.7211
aos  AP:87.57, 86.74, 78.44
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:92.9081, 88.0743, 80.6394
bev  AP:92.9499, 90.0188, 82.6217
3d   AP:92.9463, 88.2135, 82.5952
aos  AP:92.73, 87.86, 80.40

ylwhxht, 2025-04-28:
> 你好作者，我新训练了120轮，单看其中car 3d检测精度，第97轮最高，为81.7157，预训练模型是83.0870，论文是85，还是差了些，我也不知道问题出在哪儿 下面是第97轮的测试log INFO *************** EPOCH 97 EVALUATION ***************** 2025-04-27 13:54:30,580 INFO [...]

对于KITTI这么多的评估指标，我们是按照kitti benchmark标准主要关注3D、Moderate上的性能，也就是AP_R40的最后一行中间的数字，建议你按这个为标准找最优性能模型进行比较。

ylwhxht, 2025-04-28:
@czxv587 
你预训练模型的关键性能为：
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:72.8413, 70.3744, 63.3719
bev AP:88.3546, 81.4429, 75.9884
3d AP:83.0870, **76.5604**, 69.4318
aos AP:66.27, 67.13, 60.38
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:73.4453, 70.6430, 64.0664
bev AP:74.5844, 72.6442, 66.0944
3d AP:74.4895, **72.4494**, 65.8758
aos AP:70.17, 65.48, 59.03
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:93.2093, 90.0153, 82.7001
bev AP:93.3023, 90.5515, 83.1662
3d AP:93.2987, **90.5428**, 83.1074
aos AP:92.99, 89.76, 82.43
复现的97epoch关键性能为：
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:70.4151, 69.2228, 63.7739
bev AP:89.0870, 82.3368, 75.2342
3d AP:81.7157, **74.9689**, 69.4639
aos AP:63.29, 66.12, 60.84
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:73.2203, 68.7372, 63.4673
bev AP:75.7705, 71.1551, 65.8357
3d AP:75.6909, **71.0701**, 64.6195
aos AP:70.17, 64.19, 58.79
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:92.9081, 88.0743, 80.6394
bev AP:92.9499, 90.0188, 82.6217
3d AP:92.9463, **88.2135**, 82.5952
aos AP:92.73, 87.86, 80.40

预训练基本接近汇报的结果，复现的97epoch仍有差距但是相比之前复现的有了提升，建议按照表明的关键性能重新寻找最优模型

czxv587, 2025-04-29:
好的，真诚感谢作者解答，我再试试