ylwhxht/L4DR #15: gt_sampling对radar数据的径向速度的影响

wujingxian1, 2025-04-24:
您好，我想请问一下使用gt_sampling数据增强方法不会对radar数据的相对径向速度和绝对径向速度造成影响么？从训练集中采样gt，然后放入当前训练帧是怎么放置的，这个过程不会改变雷达到目标的角度么？如果角度改变了，那radar数据的径向速度就不准确了。

ylwhxht, 2025-04-24:
> 您好，我想请问一下使用gt_sampling数据增强方法不会对radar数据的相对径向速度和绝对径向速度造成影响么？从训练集中采样gt，然后放入当前训练帧是怎么放置的，这个过程不会改变雷达到目标的角度么？如果角度改变了，那radar数据的径向速度就不准确了。

我们并没有考虑这些问题，仅仅是直接使用与LiDAR处理一致的gtsampling增强。严格意义上来看你说的是对的，通过随机放置目标之后，目标内的雷达点相对自车的角度发生了变化。然而由于属于同一目标内雷达点与自车夹角差异并不大，如果我们把一个目标内的雷达点看作整体去进行放置，可以理解为是一个新的样本，我认为速度不准确的程度并不是那么大。并且我们是隐式地将其作为特征输入到网络中，而非显示基于速度的引导，我认为对于目标检测网络来说可能更关注的是是否移动而对数字可能比较忽略（当然是基于我的猜想，我并没有实验验证这一事实）。

综上，我认为：
1. 你的顾虑是正确的，我们对这方面并没有额外考虑与处理，确实可能出现不准确的现象。
2. 这个不准确的影响可能并不大，当然如果你有兴趣可以针对这一现象进行分析与设计

ylwhxht, 2025-04-24:
> > 您好，我想请问一下使用gt_sampling数据增强方法不会对radar数据的相对径向速度和绝对径向速度造成影响么？从训练集中采样gt，然后放入当前训练帧是怎么放置的，这个过程不会改变雷达到目标的角度么？如果角度改变了，那radar数据的径向速度就不准确了。
> 
> 我们并没有考虑这些问题，仅仅是直接使用与LiDAR处理一致的gtsampling增强。严格意义上来看你说的是对的，通过随机放置目标之后，目标内的雷达点相对自车的角度发生了变化。然而由于属于同一目标内雷达点与自车夹角差异并不大（因为室外大场景中大部分目标所占的比例很小），如果我们把一个目标内的雷达点看作整体去进行放置，可以理解为是一个新的样本，我认为速度不准确的程度并不是那么大。并且我们是隐式地将其作为特征输入到网络中，而非显示基于速度的引导，我认为对于目标检测网络来说可能更关注的是是否移动而对数字可能比较忽略（当然是基于我的猜想，我并没有实验验证这一事实）。
> 
> 综上，我认为：
> 
> 1. 你的顾虑是正确的，我们对这方面并没有额外考虑与处理，确实可能出现不准确的现象。
> 2. 这个不准确的影响可能并不大，当然如果你有兴趣可以针对这一现象进行分析与设计

wujingxian1, 2025-04-24:
不好意思~，麻烦您删除一下这个问题，非常感谢！