ylwhxht/L4DR #17: [Kradar Training] Missing codes in `models.backbone_3d.pointnet2`

parkie0517, 2025-05-21:
Thank you for the amazing work. 
I am trying to train the model using K-radar dataset.

However, there seems to be a problem when building the model.

Some of the files inside `K-Radar-main-repo/models/backbone_3d/pointnet2` are empty.

Specifically, the files below are empty.
https://github.com/ylwhxht/L4DR/blob/main/K-Radar-main-repo/models/backbone_3d/pointnet2/__init__.py
https://github.com/ylwhxht/L4DR/blob/main/K-Radar-main-repo/models/backbone_3d/pointnet2/pointnet2_batch/__init__.py
https://github.com/ylwhxht/L4DR/blob/main/K-Radar-main-repo/models/backbone_3d/pointnet2/pointnet2_batch/pointnet2_modules.py
https://github.com/ylwhxht/L4DR/blob/main/K-Radar-main-repo/models/backbone_3d/pointnet2/pointnet2_batch/pointnet2_utils.py

How can I fix this problem?
Do I have to install something or copy the pointnet2 code from somewhere else?

Thank you for taking the time to read my question.

ylwhxht, 2025-05-21:
> Thank you for the amazing work. I am trying to train the model using K-radar dataset.
>
> However, there seems to be a problem when building the model.
>
> Some of the files inside [...]

Hello, thank you for your interest in our work and for pointing out the problem.

This seems to be an unexpected error during the code upload process. We have re-uploaded the code for **pointnet2** part. 

Please continue to point out if you are experiencing anything else like this.

parkie0517, 2025-05-21:
Thank you for the response.

I saw the update you made. I modified 2 files. (can be seen from below)
```
K-Radar-main-repo/models/backbone_3d/pointnet2/pointnet2_batch/pointnet2_modules.py
K-Radar-main-repo/models/backbone_3d/pointnet2/pointnet2_batch/pointnet2_utils.py
```

After modifying the files, I ran the training code again.
This time, I am encountering another error.

In the file below,
```
l4dr_kradar/models/backbone_3d/pointnet2/pointnet2_batch/pointnet2_utils.py
```
If you look at line 7, it is trying to import `pointnet2_batch_cuda` file.

However, the file does not does not exist.

parkie0517, 2025-05-22:
Yes, I followed the same procedure.
I noticed that there were some files ending with ".so" in the ./ops and ./models directory. So I deleted them. Then I ran `python setup.py develop` again.
However, the "*.so" file in `./models/backbone_3d/pointnet2/pointnet2_batch/` is still just an empty file.

ylwhxht, 2025-05-22:
> Yes, I followed the same procedure. I noticed that there were some files ending with ".so" in the ./ops and ./models directory. So I deleted them. Then I ran `python setup.py develop` again. However, the "*.so" file in `./models/backbone_3d/pointnet2/pointnet2_batch/` is still just an empty file.

Hello, long time waiting. I spent some time checking the code and finally realized that this part I was copying the compiled K-Radar-main-repo/ops/pointnet2 directly into /K-Radar-main-repo/models/backbone_3d and importing it. This was not a good way to do it, resulting in the part not working after compiling directly.

We **deleted the /K-Radar-main-repo/models/backbone_3d/pointnet2** folder and **changed the K-Radar-main-repo/models/backbone_3d/pointnet2_backbone.py** code from :
```
from .pointnet2.pointnet2_batch import pointnet2_modules
from .pointnet2.pointnet2_stack import pointnet2_modules as pointnet2_modules_stack
from .pointnet2.pointnet2_stack import pointnet2_utils as pointnet2_utils_stack
```
to:
```
from ops.pointnet2.pointnet2_batch import pointnet2_modules
from ops.pointnet2.pointnet2_stack import pointnet2_modules as pointnet2_modules_stack
from ops.pointnet2.pointnet2_stack import pointnet2_utils as pointnet2_utils_stack
```

All modifications have been updated, and it should be working fine (remember to check that there is a properly compiled .so file in pointnet2 under ops). Thanks for trying, and contact us if you encounter any problems.