ylwhxht/L4DR #18: Wheel for torchvision==0.11.2+cu111 only available for Linux

chr-is-tof, 2025-05-22:
Hi, great work btw!

It seems there is no wheel to support torchvision 0.11.2 with cu111 on windows (only Linux).

Is there a workaround or other torchvision which might work?

ylwhxht, 2025-05-22:
> Hi, great work btw!
> 
> It seems there is no wheel to support torchvision 0.11.2 with cu111 on windows (only Linux).
> 
> Is there a workaround or other torchvision which might work?

Thank you for your attention to our work, we have not tried it on window, but our code has almost no requirements for torch, torchvision, cuda version, you can choose other suitable version (as long as it meets the CUDA driver corresponds to the torch version).

chr-is-tof, 2025-06-05:
Thank you! 
I used docker on WSL and it worked

closed 2025-08-24 by ylwhxht